marcelblijleven/goodwe #2: Add support for other inverters

mletenay, 2021-08-18:
Hi,
I've been developing the goodwe protocol library (mostly for the HomeAssistant) for more than a year now.
See - https://github.com/mletenay/home-assistant-goodwe-inverter
There is support for ET, ES, EM and EH inverters now.
I was for quite some time considering to extract the actual python API lib from the HomeAssistant related code and was about to do it now, when I discovered your library.

I suggest we should join forces and merge our codes to single python lib which supports all possible inverter families.

(You can reach me/us also on our discord channel discord.gg/taxywxt)

marcelblijleven, 2021-08-18:
Hi, sounds good! Will join the discord channel later this evening

marcelblijleven, 2021-08-18:
@mletenay the discord link seems to be expired

mletenay, 2021-08-19:
Weird, try this one - https://discord.gg/CAZVTDsB

mletenay, 2021-08-21:
I merged all my code into your repository, made same changes also to your code but so far did not change any of your APIs.
(During the merge I've found out the XS protocol is actually the same one as DT and D-NS inverters use.)
Please, take a look at my code changes, let me know if you dislike anything.
We should now decide how to proceed and adjust existing APIs and publish new one.
I suggest to use my Inverter classes instead of your Processors, since the inverter can not only process (single request) result, but has broader API, including methods for reading inverter settings or for setting inverter work mode etc ...

mletenay, 2021-09-02:
I think the code is getting ready for some release.
(And yes, looking at the amount of code and its changes it looks like I hijacked your repo ;-).
How are we going to proceed ?
Are you willing to add me as another maintainer to pypi so I can do it on my own ?
Anyway, feel free to comment/criticize the code and suggest any improvement...

marcelblijleven, 2021-09-02:
Hi @mletenay I haven't got the time to look into things much, busy at work and life at the moment, but if you want you can trigger the release from GitHub. It automatically builds the package and pushes it to PyPi when you create a new release from Github.

If you want, I can create it for you but feel free to try it out yourself.

closed 2021-09-07 by mletenay