marcelblijleven/goodwe #4: Multiple total counters have negative values regularly

TimelessNL, 2021-09-25:
Hi,

_First I want to say great work so far!
I was working on a [GoodWeLogger](https://github.com/jantenhove/GoodWeLogger) setup but then I came across this project.
Since I already had the WiFI dongle I figured this project/library based on UDP was worth a shot._

I've connected HA to my GW5K-DT and everything seems to work thus far.
Fortunately my `e_day` and `e_total` do work _kinda_, but they report -0.1kWh regularly:
[image]

This is also the case for the `h_total` sensor, seems like a problem with sensors reporting total values?
[image]

So my question is, is this a known issue yet?
I've enabled `debug` logging, And during a -0.1kWh moment there are no errors logged.

_Note:
This question was first posted in [#71](https://github.com/mletenay/home-assistant-goodwe-inverter/issues/71) but I moved it since the protocol logic is part of this library._

mletenay, 2021-09-26:
Well, not really sure how to cope with this.
If it is reporting non-zero (not None) values, it is definitely not the library code, so it must be the inverter which provides these odd values.
(Or in theory the library may decode wrong packet, but in such case all values reported at that point in time would be off usual values. Can you observe such correlation between various sensors ? Or is it just e_total which is off ?)

TimelessNL, 2021-09-26:
> Can you observe such correlation between various sensors ? Or is it just e_total which is off

As for now `e_total`, `e_day` and `h_total` are all negative at the same time (_maybe more_).
I have not verified `ppv` and `temperature` with the inverter at such a moment, but they look plausible during such an event.
For example this was today's `temperature` reported by the inverter:
[image]
And this was today's `e_day` reported by the inverter:
[image]
As you see `e_day` repeatedly goes to -0.1kWh shown by the blue filled graph, while `temperature` has a steady line.

I can't think of a reason why the inverter would send negative values for these kind of parameters.
My inverter is not connected with the internet (_firewalled_) but it would be strange if that's the fault.

> it is definitely not the library code, so it must be the inverter which provides these odd values.

Maybe a workaround or sanity check could be added?
For example when these parameters are reported negative return `None` instead?

mfaust78, 2021-09-27:
I haven't seen it on GoodWe but other Modbus devices, that they return -1 for "no value". the `e_total` for example is read as 4 byte signed, therefore the value on the wire would be 0xFFFFFFFF. Maybe this could be detected and as @TimelessNL suggests a `none` could be returned.

mletenay, 2021-09-27:
Yes, that sounds reasonable, at least for the 4 bytes power/energy kinds of sensors.
I'll adjust the library code to ignore such -1 values

TimelessNL, 2021-09-28:
What about the `h_total` sensor? That one seems to have `no value` results as well.

mfaust78, 2021-10-03:
Had some issues with my utilities (they remote-shutdown my inverter, which then went to "Off-Grid" mode. Took me a while to find the root cause.
During the process I restarted the inverter a few times and during two restarts, the following total values were read as "0", which screwed up the statistics for the "Energy" Panel.
Affected totals:
- `sensor.goodwe_e_total`
- `sensor.goodwe_e_bat_discharge_total`
- `sensor.goodwe_e_bat_charge_total`
- `sensor.goodwe_meter_e_total_imp`
- `sensor.goodwe_meter_e_total_exp`
- `sensor.goodwe_e_total`

As these values are ever increasing, would it be possible to ignore "0" readings?

mletenay, 2021-10-15:
-1 handling value was implemented, filtering 0 values for total_increasing sensors was done in HA code.

closed 2021-10-15 by mletenay