marcelblijleven/goodwe #8: setup.cfg references a file "VERSION"

onkelbeh, 2022-01-28:
which is missing in Pypi's SDIST archive.

onkelbeh, 2022-01-28:
Hope you don't mind I use your component for [Home Assistant for Gentoo](https://github.com/onkelbeh/HomeAssistantRepository)

The (fixed) Ebuild was committed in https://github.com/onkelbeh/HomeAssistantRepository/commit/0c3143468c0504a330944fa2eeb71b824a22b086

Please take a look at the neccesary fixes:
```bash
src_prepare() {
	echo ${PV} > VERSION || die
	echo "[options.packages.find]" >> setup.cfg || die
	echo "exclude = tests*" >> setup.cfg || die
	default
}
```
Filed a [PR for the tests exclusion](https://github.com/marcelblijleven/goodwe/pull/7)
Please add an automatism for the 'VERSIN' File.

marcelblijleven, 2022-03-14:
Closed by #7

closed 2022-03-14 by marcelblijleven