marcelblijleven/goodwe #10: pv mode 1 and 2 switched?

nickelbert, 2022-03-05:
PV1 has 0W and PV2 has 17W but pv mode shows PV1 producing power and PV2 not connected

Model: Coodwe GW8K-ET

[image]

lyricnz, 2022-10-23:
Which model inverter?  Can you provide the debug output?  logging.basicConfig(level=logging.DEBUG)

gowima, 2024-01-15:
Same for me forGW29K9-ET inverter with current (2024-01-14) version of goodwe module. Pypi module 0.2.33 with byte address mode was ok for me.

The GW29K9-ET inverter has 3 MPPT. I could watch the above decribed behavior for PV3 and PV4. 

This seems to a problem with byte order (big endian vs little endian) in the current implementation using register address mode. Have a look at my fork of goodwe at gowima/goodwe (be aware this is still work in progress).

mletenay, 2024-01-16:
Fixed in 88a1ff7e4560f4034528304a744dfe3fdc7d51da
Thanks @gowima

closed 2024-01-16 by mletenay