marcelblijleven/goodwe #15: General Electric whitelabelled Goodwe's have different serial numbers

StephenPritchard, 2022-06-26:
Hi there,

Thanks for a great repo!

I have a General Electric (rebranded) goodwe inverter, and the serial number format is different to normal Goodwe inverters.

As a result, this code in the DT inverter class does not work (my single phase, 3-pv-string inverter does not contain either DSN or MSU:
```
if "DSN" in self.serial_number or "MSU" in self.serial_number:
            # this is single phase inverter, filter out all L2 and L3 sensors
            self._sensors = tuple(filter(self._is_not_3phase_sensor, self.__all_sensors))

        if "MSU" in self.serial_number:
            # this is 3 PV strings inverter, keep all sensors
            pass
        else:
            # this is only 2 PV strings inverter
            self._sensors = tuple(filter(self._is_not_pv3_sensor, self._sensors))
        pass
```

I don't know how to work out phase/pv string information from the general electric serial number, so in my case, I just cloned the repo and hard-coded these for my case.

How did you go about working this out for Goodwe?

mletenay, 2022-07-06:
What is your serial number format, resp. what is the exact name/model ?

mletenay, 2022-08-09:
Well,
I was able to deduct that PSB is model name for 2MPPT 1phase models, and PSC for 3MPPT 1phase models
(see the model page url https://www.gesolarinverter.com/PSB.html).
Unfortunately, the 2MPPT 3phase model page does not follow that naming convention, so unless someone reports the serial nr, those will not be supported.
But I did add the recognition of PSB and PSC models, so the next release (0.2.19) should work for you out of the box.

closed 2022-08-09 by mletenay