marcelblijleven/goodwe #19: GoodWe firmware version breaks the es config after update

NielZeeman, 2022-07-06:
My inverter was updated to version 2424G
This seems to break on the following line: (es.py:160)
self.arm_sw_version = int(self.arm_version[4], base=36)

mletenay, 2022-07-06:
I assume it failed on line 160, but the previous version with base=16.
The base=36 is the fixed done here https://github.com/marcelblijleven/goodwe/commit/327c7803e8415baeb4b6252431db91e1fc6f2fb3 and I'm going to make a release with it soon.

closed 2022-07-06 by mletenay

NielZeeman, 2022-07-07:
Great. Thanks a lot!!

DarthV02, 2022-07-27:
> My inverter was updated to version 2424G This seems to break on the following line: (es.py:160) self.arm_sw_version = int(self.arm_version[4], base=36)

Hi @NielZeeman, as you will see on the chat que, I also commented on this issue, as i have the same problem you posted. 
I understand (from their comments) that the programmers are working on it, but the goodwe inverter official integration for Home Assistant Supervised, isn't still working for me... the base=16 issue keeps popping up when traying to install it. 

Just to check whether i am doing something wrong, would you please let me know if you have it up and running?  
Thanks in advance! :)

Aitor.

starkillerOG, 2022-08-12:
The issue schould have been fixed with this PR: https://github.com/home-assistant/core/pull/75615
This fix is included in HomeAssistant version 2022.08.x.
Sorry for the late replay, it is vacation time...

DarthV02, 2022-08-12:
Come on! Nothing to apologize for, we really appreciate the amazing job you do!
Many thanks for the response and the fixing!
Enjoy the rest of the summer you all. 😊