marcelblijleven/goodwe #21: Support for ESA models

gamosoft, 2022-08-11:
Hi there, I have a GW5048ESA and I'm trying to get the data on my own.
I'm able to figure out a handful of values by trial and error, but I don't know the rest (such as operating mode, battery status mode, etc). :-(

[image]

Any plans to add support for this, or at least know where I can find some info?

Thank you!

mletenay, 2022-08-12:
Is there actually any difference between ESU and ESA models ?
In explicit inverter mapping the ESA is not present yet, but doesn;t the automatic protocol discovery identify the inverter es ES family?
And no, there are no public documents describing the protocols out there AFAIK.

gamosoft, 2022-08-12:
Hi there, thanks for answering! :-)

Not sure about the ESU but what I mean is some kind of documentation like this (that I found through some forums but don't remember now where):

http://yamasun.com.tw/upload/F_20170313191367UrC8jo.PDF

Following the PDF info above I built a C# small tool and I can query for the data on my ESA model, get PV voltage, current, total power... etc but I'm missing a lot of info which I don't know what it means, excluding message headers and checksum, the actual data I get back is this:

[image]

I don't know what data to get battery status, inverter status, configuration parameters... having in mind that sometimes you may need to read1, 2 or 4 bytes out of the response (or maybe even odd numbers).

My aim is to have some kind of info like this one that someone has compiled for the ES model (not applicable to ESA model as you can see):

https://powerforum.co.za/topic/6931-goodwe-5048es-protocol-reverse-engineering/?do=findComment&comment=83134

I will look into the ESU in the meantime and see if that helps.

Kind regards.

mletenay, 2022-08-12:
Err, I'm not sure I understand what you need here.
Simple question: 
Does this python library work properly on "ESA" model (by implicit or explicit inverter type selection) or not ?

gamosoft, 2022-08-12:
Hm I think I may have a clue how to continue.
I was following the PDF documentation to send message with request [01, 01], which is in fact returning data but *maybe* it's specific to the ESA model.

By looking at the source code here:
https://github.com/marcelblijleven/goodwe/blob/3675fa6c28e4ac4b4bef5b7664c897f31d6f4ced/goodwe/es.py

I can see the actual request message is [01, 06], so the response *seems* to match the one specified in the page source so I'm going to give that a try. ;-)

Is there some place that specifies the different message requests and what they do?

For instance 01,06 to get data, 01,02 to get inverter info, 01,03 to get settings info...

Thanks for your help!

mletenay, 2022-08-12:
... at the top of that class:
    _READ_DEVICE_VERSION_INFO: ProtocolCommand = Aa55ProtocolCommand("010200", "0182")
    _READ_DEVICE_RUNNING_DATA: ProtocolCommand = Aa55ProtocolCommand("010600", "0186")
    _READ_DEVICE_SETTINGS_DATA: ProtocolCommand = Aa55ProtocolCommand("010900", "0189")
   
But to my question - does or does not the library work with ESA model ?

gamosoft, 2022-08-12:
yes i hadn't looked into the actual code and i'm afraid i haven't tested it since I don't use python :-(
looks like there are several more commands that can be sent, and some seem redundant (as i said, 0106 seems to be getting similar data as 0101, but in a slightly different order)

in any case, i think i have all the info i need by taking a look at the source here so i can adapt to my app:

[image]

thanks for all your help!

closed 2022-08-12 by gamosoft