marcelblijleven/goodwe #22: Received invalid response on GW10K-BT, FW: 02041-22-S00

MitchVL, 2022-08-22:
Hello,

Connecting to my GW10K-BT inverter is possible, and identifies the following:
Identified inverter
- Model: GW10K-BT
- SerialNr: XXXXXXXXXXXX
- Version: 04029-06-S11
- Firmware: 02041-22-S00

The inverter firmware was updated on 16/8 by goodwe support, so it should be the latest.

When I run read_runtime_data() something goes wrong.
Below you can find the log after the call is made.
Changing the amount of retries to 10000, will give a correct result after a random amount of tries.

2022-08-22 22:18:12,952 - goodwe.protocol - DEBUG - Sending: f703891c007d7ae7
2022-08-22 22:18:12,980 - goodwe.modbus - DEBUG - Response is too short: 54, expected 257.
2022-08-22 22:18:12,981 - goodwe.protocol - DEBUG - Received invalid response: aa55f703fa16081616113800000000000000000000000000000000000000000000000000000000000000000000000008a80010138800
2022-08-22 22:18:12,985 - goodwe.protocol - DEBUG - Sending: f703891c007d7ae7 - retry #1/3
2022-08-22 22:18:13,014 - goodwe.modbus - DEBUG - Response is too short: 42, expected 257.
2022-08-22 22:18:13,015 - goodwe.protocol - DEBUG - Received invalid response: aa55f703fa16081616113800000000000000000000000000000000000000000000000000000000000000
2022-08-22 22:18:13,019 - goodwe.protocol - DEBUG - Sending: f703891c007d7ae7 - retry #2/3
2022-08-22 22:18:13,048 - goodwe.modbus - DEBUG - Response is too short: 42, expected 257.
2022-08-22 22:18:13,049 - goodwe.protocol - DEBUG - Received invalid response: aa55f703fa16081616113800000000000000000000000000000000000000000000000000000000000000
2022-08-22 22:18:13,050 - goodwe.protocol - DEBUG - Sending: f703891c007d7ae7 - retry #3/3
2022-08-22 22:18:13,982 - goodwe.protocol - DEBUG - Max number of retries (3) reached, request f703891c007d7ae7 failed.

Thank you !

mletenay, 2022-08-24:
Wifi or LAN module ?
Since there were reports LANs are sometimes less reliable than Wifi - https://github.com/mletenay/home-assistant-goodwe-inverter/issues/116
Anyway, I'm afraid there's no help (or explanation) there.
It seems the UDP packets arrive to your HA instance truncated/incomplete. 
It may be network or HW issue, we really do not know and the only known remedy is to increase re-try counts (or to fix net/hw issue if there is any).

MitchVL, 2022-08-24:
Thank you for your quick reply.

We have the lan module installed.

I have also been in contact with someone who creates bms for goodwe inverters, and he mentioned that if there is something controlling the inverter through another connection method, you probably also see that communication on Lan.

We have an intelligent system controlling our inverter through serial connection. So that means that the packets that we see on Lan, are probably the responses that are also sent on the serial connection.

Unfortunately that means that I can not get any information out of the goodwe inverter on command.
The only thing I can do is sniff the packets, looking for data.

Thank you for your good work and assistance

mletenay, 2022-08-25:
I'd love to help LAN module users, unfortunately I do not know how at the moment.
It smells like if the inverter was sending broadcast? UDP packets ?
If you have any idea how to filter those "noise" packets, I'll try to implement it.

lyricnz, 2022-10-22:
FWIW I also have a lan module (on a GW20KAU-DT) and can tcpdump etc if that would help.  If necessary I can insert a switch with a monitor port to capture all.

NicolaVerbeeck, 2022-10-29:
I did some digging on my own system since I get the same issue (truncated data). I wonder if this is caused by another system requesting data from the inverter and that causes its network buffer to be truncated. I tried to request smaller chunks in a test scenario (eg: chunks of 8 bytes to build up the 257 long message) and this seems to work a lot more reliable . If I increase the chunk size to say 40 bytes, I already see truncated UDP packets.

Will investigate later by removing the other system from the equation and see if that solves the truncated data issue.

lyricnz, 2022-10-29:
@NicolaVerbeeck are you running >1 system talking to the inverter? (maybe two copies of HA?).  Or could it be the inverter calling home to goodwe/semsportal?

NicolaVerbeeck, 2022-10-29:
I'm pretty sure there is a 'power usage optimizer module' RPi from the installer also communicating over UDP with the inverter. Will disconnect it later and see if that clears this up. Then at least we have an idea what could be causing this (low network buffer). It would not be the first time I ran in into this buffer issue, even at very large clients

lyricnz, 2022-10-29:
There’s a know issue where this module can interfere with inverter talking to semsportal (completely kills it for me). Maybe that’s the same issue

NicolaVerbeeck, 2022-10-29:
Unfortunately unplugging this additional device did not yield better results on my GW5000-EH. I still never get a full 257 message in one go. The longest I have seen so far is 192

closed 2024-02-14 by mletenay