marcelblijleven/goodwe #23: Incorrect values for ES series?

gamosoft, 2022-08-25:
Hello again, as I mentioned in a previous discussion your code greatly helps me move forward reading the values of the UDP messages using a C# application, I already had a few figured out (the easy ones) by just comparing the values but others were simply impossible for me to decipher until I stumbled upon your code. :-)

As a summary I have a ESA model inverter and the misunderstanding I was having was between the message to send, since tech support sent me a PDF document where they describe some values and the message to send as [01,01] (response [01,81]).

However looking at your code I saw you send [01,06] (response [01,86]) so the bytes obviously mean different things and thanks to your descriptions I changed my message so now I can get most of the stuff I need to replicate the chart seen in SEMS portal easily:

[image]

Where did you find the values for the response message and their meaning?
I'm asking because I'm having a slight discrepancy which I'm not sure how to figure out (or maybe I'm missing some calculation).

When I get to see the total PV generated (for example) during the inverter lifetime in response [01,86] as per your code it's read from bytes [59 .. 62]:

```code
Energy4("e_total", 59, "Total PV Generation", Kind.PV),
```

But if I compare this value to what I see on SEMS they're slightly off:

[image]

Yes, I know some other values are off here due to the fact SEMS updates every 5 mins, but you get the point. ;-)

However, if I query the inverter telemetry for response [01,81] on bytes [54 .. 55] I do get the value shown in SEMS:

[image]

Do you know where can the disconnect be or what am I missing in the calculations? Same happens for today's generation, amount bought/sold today etc...

That's why I was asking also where you got those specs from or just by trial and error? :-)

Thanks again.
Kind regards.

mletenay, 2022-08-25:
I did not have any specs, everything was reverse engineered and trial/error.

gamosoft, 2022-08-26:
Ok thanks, I guess I'll probably have to issue several commands to get the values I might need so they match up those in SEMS (assuming SEMS is accurate which I'm starting to believe not so much). ;-)

closed 2022-08-26 by gamosoft