marcelblijleven/goodwe #24: Sorry - fundamental question about how to setup?

lyricnz, 2022-09-11:
Hi. Sorry this is a dumb question.  I have a GOODWE GW-GW20KAU-DT(G2)

Which is connected to my wifi, and uploading data to sems portal fine.  However, I cannot connect to it using this tool/library (nor see any UDP traffic from the device).  Is there something I need to do in order to set this up on the inverter?  Would love to grab data directly, and not use the portal.

lyricnz, 2022-09-11:
Is it, perhaps, only available when the inverter is running it's own wifi network (and you are connected to it), rather than when it's connected to existing wifi?

lyricnz, 2022-09-12:
OK, I connected to the "Solar-WiFi" network, and ran the test/query program, and it worked perfectly!  Is this expected?  Do I need to enable something to make this work via regular wifi?

mletenay, 2022-09-12:
No, there is nothing necessary (or even possible) to be configured.
It should work out of the box.
If you cannot reach inverter from your wifi, I would suspect something wrong with your network setup.

lyricnz, 2022-09-12:
I suspect you're right. It was on my guest-wifi, and so was my raspberry pi.  Moving it to the real wifi made it work fine.

closed 2022-09-12 by lyricnz