marcelblijleven/goodwe #25: 1500 XS, FW 1.53.50 not working

kbfifi, 2022-10-20:
Hello Marcel,

I tried several Py test scripts. None of them will respond. I was able to send this:
echo "WIFIKIT-214028-READ" | nc -v -u -n -b 192.168.178.130 48899

I got the the serial numer back. I understand from your comments this is only the Wifi part talking back. Nothing comes back from port 8899. I was able to setup the SEMS portal. (SolarGo is NOT working and says the function is not supported by this device) I made a tcpdump and concluded that 8899 or even 48899 was not even used by SEMS. Could it be that SEMS is using a total different protocol? 

As I believe I have a fairly new firmware I'm afraid Goodwe has changed something that breaks the support for port  8899. Any suggestions?

mletenay, 2022-10-21:
AFAIK, inverter is pushing data to SEMS portal, so there is no listening port there.
As for the firmware, as mentioned in description, there seem to be 2 firmwares - "inverter" and "arm".
It's the arm firmware which tends to be old and not exposing UDP protocol.

kbfifi, 2022-10-21:
How can I tell which of the firmwares I have? I guess both support the UDP protocol if  arm type is of recent type. Port 8899 IS open but it won anwer anything. I also noticed the base adress to be set to 247 (=F7) on the LCD screen. Looks fine too. It should work. 

Do you know if SolarGo also uses the UDP protocol? I read in the app description it is only supported by Inverters without LCD screen. I got the impression that other people got it to work with the 1500 XS as well.

mletenay, 2022-10-21:
Yes, both PvMaster and SolarGo use UDP protocol exposed on port 8899, just different protocol format/language.
So the usual rule is - the library works when SolarGo works.
If I remember correctly, there's a mention in some ticket about users who asked goodwe support to upgrade their (arm) firmwares and after that it started to work.

closed 2023-01-31 by mletenay