marcelblijleven/goodwe #27: Error message about 149 != 149 is confusing

lyricnz, 2022-10-23:
As seen in #20 the following logs were omitted:

```
DEBUG:goodwe.protocol:Sending: aa55c07f0102000241
DEBUG:goodwe.protocol:Response has unexpected length: 149, expected 149.
DEBUG:goodwe.protocol:Received invalid response: aa557fc001868c0ec5000c000000000100020f019f0050010400360064006400001c001c6403000001097800180001138b0100000016ff02138d00020160000000000000048e0000016d0011003d00000f75ff0201003000000000010000febf100008000000160a160d0e25000000000000000000000000000000000000000000000000000004e7000004bf020000000000001128
DEBUG:goodwe.protocol:Sending: aa55c07f0102000241 - retry #1/3
DEBUG:goodwe.protocol:Received: aa557fc001824d3130313043475735303030532d42502331300000000000000000000000000039353030304250533232335730313336333630303431302d30343030312d3130093431302d30323033342d313502100e
```

This includes the nonsensical message "has unexpected length: 149, expected 149".

Given that 149 is an odd number this can only come from the following code in goodwe/protocol.py

```
        if (
                len(data) <= 8
                or len(data) != data[6] + 9
                or (response_type and int(response_type, 16) != int.from_bytes(data[4:6], byteorder="big", signed=True))
        ):
            logger.debug("Response has unexpected length: %d, expected %d.", len(data), data[6] + 9)
            return False
```

It looks like the third condition in the "if" is not accurately described by the error message.

lyricnz, 2022-10-23:
https://github.com/marcelblijleven/goodwe/pull/28

mletenay, 2022-10-24:
Thanks for fix, your PR was just merged.
Glad to see another developer, especially with different inverter model.

closed 2022-10-24 by mletenay