marcelblijleven/goodwe #31: Peak shaving is missing from the GoodWe inverter operation mode

MrOzzOnline, 2022-11-16:
GoodWe has added a peak shaving mode in ARM version 22 for the ET series. See: https://community.goodwe.com/Forum/details/2c9180827a7b69a0017a94a74d0c0281

It is also mentioned in the manual.

This option is missing in Home Assistant

This feature was implemented for the Belgian market for example. There will be a peak tariff starting from 1/1/2023

Goodwe's implementation of the peak shaving might not be the most ideal, though it can be useful when it's combined with some smart logic, for example: connect car >> switch mode >> charge battery taking into account the maximum peak >> charge EV and stop when battery is depleted, repeat until car is full. >> switch back to general mode.

So, it's an important feature.

MrOzzOnline, 2022-11-18:
@marcelblijleven could you have a look at this issue?

mletenay, 2023-02-18:
Peak shaving mode was added, in HA it will be part of 2023.3 or HACS 0.9.9.7

closed 2023-02-18 by mletenay