marcelblijleven/goodwe #37: Immediate charge to Battery is missing

XanderF, 2023-01-24:
For GW3600S-BP and GW5000S-BP models (probably other models as well) it can be useful to charge the battery on command. 

The PV master app contains the option 'Immediate charge to Battery' which is used to charge the battery immediately. GoodWe's description is 'Set for battery fast charging. It will only take effect once'. There is also a related option to stop charging: 'Charge cut-off SOC'. The battery charging will stop once the battery SOC reaches the Charge cut-off SOC.

It would be very helpful if this/these option(s) would be accessible from Home Assistant. It would allow people to charge the battery when there's no sun, but strong winds, thus a low hourly energy price.

antonclaeys, 2023-02-08:
Have you found out something about this in the mean time? I'm very interested in this to write something for trading on the energy market with these inverters and batteries

XanderF, 2023-02-08:
A couple of days ago I read that the experimental version contains the 'fast charging' option. However it doesn't get mentioned in the readme anymore unfortunately. See https://github.com/mletenay/home-assistant-goodwe-inverter

The battery-only inverter I have, GW3600S-BP, is not very well supported yet by this integration. For example, DOD is not doing anything when changed from HA. And the state in HA ins't updated if it is changed in the PV Master app.

I'm soon going to add the requests and responses of my inverter when changing the 'fast charge' option. Maybe that helps someone in adding this feature to this project.

mletenay, 2023-02-08:
Captured requests/responses would help a lot, I'll try to figure out the rest.

XanderF, 2023-02-24:
There's some news on this issue, but I'm stuck. Not sure how to proceed. Here's what I did.

- Captured UDP traffic between the GW3600S-BP and my iPhone using Wireshark on a Mac and the PV Master app on my iPhone. All traffic starts with the known AA55 sequence. I haven't posted the results, because I needed to figure out if my inverter serial was anywhere within the result. Haven't had time to do that.
- I've talked to GoodWe support and they said I should not use PV master anymore. It will not be updated. Use Solar Go instead, because that's now a universal app for all types of inverters. Tested the app and it does work. Looks pretty similar to PV master. Ok, so now I need to capture the traffic again using Solor Go.
- There's a document titled "Protocol for Solar Inverter Family" (source https://github.com/tkubec/GoodWe/wiki/AA55-protocol) which contains info about the AA55 protocol, but it's from 2017. I was wondering if I could receive an updated version from GoodWe, so I've asked them and specifically asked for info about the SBP family. Got an updated version, but it does not describe the AA55 protocol, but the Modbus protocol. However on the 4th page it says "This is a map document of standard MODBUS RTU protocol for only GoodWe energy storage inverters compatible with LV battery – ES, EM, SBP,ESA,BP series" :)

So now I'm confused. My SBP inverter can communicate using Modbus as well? I did a quick capture using SolarGo and it communicates with the AA55 protocol. I want to figure out how I can communicate over wifi using the Modbus protocol. Capture the data and post it here. Is that the way to go, or I'm I on the wrong track? Otherwise I'll just post the AA55 communication.

mletenay, 2023-02-24:
Can we move the protocol discussion to the discord (https://discord.gg/TaXyWXT) ?
It's more convenient there ..

iroger, 2024-01-28:
Hi chiming in here, just installed a GW6000-ES-20 ("ESN" version = ES Gen 2) it looks like they have lower ARM version numbering compared to the other units. I'm running below versions after updating in Solar Go app:
Identified inverter
- Model: GW6000ES20
- SerialNr: 56000ESNxxxxxxxx
- Rated power: 6050
- A/C output type: 0
- Firmware: 04048-06-S06
- ARM firmware: 02020-08-S01
- Modbus version: 121
- DSP1 version: 6
- DSP2 version: 6
- DSP svn version: 6606
- Arm version: 8
- ARM svn version: 362

With this ARM version I DO have: 
- Self-use mode with on-grid and off-grid DOD + under advanced settings: Back-up mode / Economic Mode / Smart charging
- Peak shaving mode
- Load control

However in the goodwe/et.py code I see some hard checks on ARM >19 and >22 which disables these functions.
It looks like these are supported in my ARM version 8. (See attached)

[image]
[image]
[image]