marcelblijleven/goodwe #38: Inverters With 4 MPPTs

melvanderwal, 2023-01-26:
The GoodWe Home Assistant integration works very well with my GE-branded GoodWe GEH inverter.
However, my inverter has four PV strings (MPPTs), and the integration only reports sensors for two strings (i.e., sensor.pv1_power, sensor.pv2_power). The goodwe.pv_power sensor only totals PV1 and PV2. It would be fantastic to have support for 4-string GoodWe inverters.

All four PV strings are available through the GoodWe SEMS API.

I am currently using the experimental version of the HA integration.

I see in the ET class that the code is there but commented out - I'm guessing that you have no way to test it.  

Model: GEH10-1U-10 (the device misreported in HA as GEH10-1U-1; I suspect it's truncated at 10 chars)
S/N: 9010KHSB21AW0043
Firmware:  43-2S002-S10 (02041-23-S00)

I have attached some debug level HA log entries.   I'd be happy to test the code on your behalf, if you'd like - if I can help in any way, please let me know.

[goodwe_log.txt](https://github.com/marcelblijleven/goodwe/files/10505414/goodwe_log.txt)

mletenay, 2023-01-26:
The 4 MPTT inverters support should be implemented in v0.2.21.
I have also created new release of HACS' version of the HA plugin (https://github.com/mletenay/home-assistant-goodwe-inverter).
I suggest you to install that one and test it before I will push the change to HA core.
Just to be sure it works as expect.

closed 2023-01-26 by mletenay

melvanderwal, 2023-01-27:
Excellent!   I have only verified the PV Power entries for the individual MPPTs and the total PV power (see attached image).  I will check the other entities, and if I find any problems I will let you know.

Thank you for updating this so quickly, I appreciate it!  I started my Home Assistant journey last week, and am loving it.
[image]

melvanderwal, 2023-01-27:
Hi @mletenay, just letting you know that the rest of the entities appear to be correct (see screenshot below).

The watts are correct within a couple of watts (watts = volts x amps).  The PV Power is the correct sum of the individual PV power measurements.  Today's PV generation matches what I see in SEMS.  

Thanks again for making the adjustment to the code so quickly!
[image]