marcelblijleven/goodwe #41: Improve support for MS models

sjtrny, 2023-02-11:
Hi, I recently had a GW6000-MS inverter installed which is single phase and has 3 MPPTs. I noticed in Home Assistant the following problems:

1. Sensors for three phases are shown.
2. Only `pv1` and `pv2` strings are shown.

I've tracked the issue down to the following lines, which miscategorises my inverter. I've locally tested a fix by adding in another serial sub-string `'MST'` and the problem is resolved.

https://github.com/marcelblijleven/goodwe/blob/33d90cb54e4546d8d7eacae769bd10ec7749e6ea/goodwe/dt.py#L147-L156

sjtrny, 2023-02-11:
May I suggest that these checks use `self.model_name` if it is available? It seems like a much more reliable option.

mletenay, 2023-02-11:
Fixed in v0.2.25

closed 2023-02-11 by mletenay