marcelblijleven/goodwe #42: Standby mode?

liebefuroscar, 2023-02-25:
I´m using my Goodwe EM5048 with DIY battery. 
I have an electricity tariff with cheap hours and expensive hours, in the VERY cheap ones I am interested in charging the battery and in the VERY expensive ones I am interested in discharging the battery. But there are some hours that are not cheap enough to upload and not expensive enough to download. I have tried setting ECO CHARGE/DISCHARGE to 0% but it charges/discharges uncontrollably. The closest thing to what I need is to put ecomode at 1% and it only draws 1A. but is there the possibility of a "standby mode" that the battery current stays at 0A?

mletenay, 2023-02-25:
AFAIK, the "backup mode" is exactly for that, it functions like big UPS then. 
As long as there is grid power, the battery is just being charged or standby.
It discharges only when grid fails.

closed 2023-02-25 by mletenay

liebefuroscar, 2023-02-25:
[image]
[image]
When I put backup mode download the battery towards the consumption of my house

mletenay, 2023-02-27:
Try to change the mode from PvMaster or SolarGo.
Maybe the problem from #164 is affecting you.