marcelblijleven/goodwe #43: Use of non default port

m-jarzebowski, 2023-03-02:
Hi, thanks for you work.
I'm trying to set up connection to my inverter in a different network for separation. The inverter is over NAT. The connection is forwarded from port 50 to 8899. I saw that some time ago support for non-standard port was dropped. I'm wondering if it is possible to bring port configuration back - would be great.

mletenay, 2023-03-02:
The port configuration was dropped a long time ago, since the 8899 is actually hard-wired to inverters.
In the very early implementations I thought that can be configured, so it was also exposed in library, but it is not.
I do not think it is worth making the code/configuration unnecessarily complex for such extremely rare usage like yours, I'm not really inclined to do so.
I guess you should be able to configure your network segment in a way you can use same port in both...

closed 2023-03-02 by mletenay

m-jarzebowski, 2023-03-03:
OK, yes I'm able to configure network lik you say, but that is not only one device. From security perspective inverters shouldn't be in the same network as client. So if you have two inverters it is not possible to use the lib for both. Thanks for answering.