marcelblijleven/goodwe #44: How to change General-Eco mode

Dupl3xx, 2023-03-03:
Hi, reading the values from the inverter works perfectly for me. Could you please advise me, if I want to change the MODE via python, How do I achieve this? I can't seem to figure it out. There is only a sample to read the values. But I would need to write something to the inverter as well.

Thank you

mletenay, 2023-03-05:
Look at the Inverter class, there are docs for all public methods.
There is set_operation_mode for changing the inverter mode.
Be aware of the usual disclaimer - writing most likely works, but if you do something wrong it can mess your inverter. 
Use with care and on your own risk.

closed 2023-03-05 by mletenay

Dupl3xx, 2023-03-06:
I've tried but I just can't get it to work.
I found this, but I can't get it to work. Could you please post a direct code sample (like the one for reading the states) on how to change the MODE.  

# ---------------------------
# Set inverter operation mode
# ---------------------------
# print(asyncio.run(inverter.get_operation_mode()))
# asyncio.run(inverter.set_operation_mode(goodwe.OperationMode.BACKUP))
# print(asyncio.run(inverter.get_operation_mode()))

Thank you

mletenay, 2023-03-06:
That is exactly the sample code.

Dupl3xx, 2023-03-06:
Yes, i try it but its not work Could you please write me this script ? Very thank you.

S474N, 2023-03-06:
Hi, sorry for bothering you. We are beginners and would like to control Goodwe from Nodered. We can read parameters from Goodwe, but cannot set it. Can you help us with generating buffer query, please?