marcelblijleven/goodwe #46: daily statistics for ET inverters

mdughera, 2023-04-13:
I may be wrong but there is something strange in the values for daily statistics for ET inverters. 
I cannot compare the values obtained from this library from those seen in the sems portal. 

Looking at the register config, the sequence is not simmetric:
        Energy4("e_total", 182, "Total PV Generation", Kind.PV),
        Energy4("e_day", 186, "Today's PV Generation", Kind.PV),
        Energy4("e_total_exp", 190, "Total Energy (export)", Kind.AC),
        Long("h_total", 194, "Hours Total", "h", Kind.PV),
        Energy("e_day_exp", 198, "Today Energy (export)", Kind.AC),
        Energy4("e_total_imp", 200, "Total Energy (import)", Kind.AC),
        Energy("e_day_imp", 204, "Today Energy (import)", Kind.AC),

"Today's PV Generation" is Energy4 (4 bytes) while "Today Energy (export)" and "Today Energy (import)" are Energy (2 bytes).
Also "Hours Total" appears between the export measures.

Obviously the asymmetry is not necessary an issue, but since I obtain strange results I suspect that something should be fixed. 

TIA

mletenay, 2023-04-13:
Nope.
These sensors are pointing at correct modbus registers and exactly match the inverter protocol.

See values from my ~2.5 years old inverter:
```e_total                       :	Total PV Generation = 18508.9 kWh
e_day                         :	Today's PV Generation = 6.9 kWh
e_total_exp                   :	Total Energy (export) = 16389.7 kWh
h_total                       :	Hours Total           = 23623 h
e_day_exp                     :	Today Energy (export) = 6.7 kWh
e_total_imp                   :	Total Energy (import) = 67.5 kWh
e_day_imp                     :	Today Energy (import) = 0.0 kWh
```

mdughera, 2023-04-19:
Hi Marcel,

I see that e_day_imp is zero.
I cannot tell if it is correct in your config, but I see zero also in my ET inverter, even though the daily import in SEMS is not zero.
This is the reason I made the request.

Ciao,

Marco

Marco Dughera
HR & Sourcing Italy

Reply
Via Cardinal Massaia, 83
10147 - Torino - ITALY
phone: +39 011 29100
www.reply.com

[Reply]

mletenay, 2023-04-19:
As mention in comment above, don't be confused with the various "imports" and compare the same values between inverter and SEMS. 
The sensor e_day_imp is not amount of energy imported into household from the grid. 
That energy is measured by the "meter" sensors (which report values measured by CT clamps).
AFAIK, the e_day_imp is the energy imported (from grid) into the inverter system (mostly battery) - i.e. when using EcoMode charge from grid.