marcelblijleven/goodwe #49: Conflicting use with Goodwe Sems Portal

philwild, 2023-04-26:
I don't know why, but when I setup a 1 minute sample from my Goodwe inverter using this library, the inverter stops reporting up to the seems portal until after I terminate my local monitoring.

philwild, 2023-04-26:
This is the code I've implemented to read from the inverter every 60 seconds...

while True:
    result = asyncio.run(get_runtime_data())
    print(result)
#    status = send_to_splunk(splunkhost, splunktoken, result)
    time.sleep(60)

You'll notice the PV and Load lines go flat after I start running the above code...

[image]

mletenay, 2023-05-04:
This problem has been reported by a small fragment of users and there is no known explanation of it.
Vast majority of users use it with 30s interval without problems, I do use 10s polling for years with no issues.

We can/should collect affected inverter(+wifi) models and firmware versions to find out if it is HW/SW specific, but unless someone provides some explanation, there's not much to do here ...

oziee, 2023-05-30:
yes.. for my SBP too
also if the HA add-on is running and I open the inverter web interface admin page, if I set any wifi etc values and save, they are not applied.. however if I disable the HA integration and on the inverter page click save now, the settings are saved

this page :)
[image]

I still have not worked out why..

stevoh6, 2023-05-30:
I have similar issue with GW4K-DT

HealthyFridge, 2023-07-22:
Same problem with a GW5000-SDT-20 (firmware 16.16.18) with Wifi-module (firmware version V1.1.3.10).

d-uhlig, 2023-09-01:
Has anyone found a solution? I have the same problem with a GW10KT-DT.
Only from an interval of 180 seconds it reports again to SEMS.

Jan-Antos, 2023-10-20:
I have GW15K-ET, same problem. Even 2-minute update interval will result in "server connection" diode flashing and no data being transmitted. It evens block the ability of goodwe to upgrade firmware because they see the inverter as offline. If I stop HA, the light stops flashing and inverter is connected. I am running now with 5 minutes and SEMS portal  is updated ok.

TomEkk33, 2023-10-20:
@Jan-Antos 
GE10k-ET and lower powers are fully supported by HA
Please note that GW15k-ET is a different branch of inverters with separate FW (ETT), comparing to GE10k-ET and lower powers FW (ET_BT).

It's ET name suggests that it is the same family, but all ET > 10kW are **not** the same as GW(5,8,10)k-ET models (PLUS+).
**Try to check if you have the latest FW version. If not ask GW support to update.**
**I'm not sure if ETT version (ET > 10kW) are fully supported by HA.**

I have GW10k-ET (PLUS+) model with latest Oct-2023 FW (ET_BT). Previous versions also didn't have issues with SEMS.
A 10s pooling interval by HA doesn't block SEMS. Sometimes SolarGo needs few more tries to connect.

In case of GW10k-ET inverter updates are online (by GW local support) or by Bluetooth dongle 
(replacing the WiFi dongle) and SolarGo app.

Kveri, 2023-10-31:
> This problem has been reported by a small fragment of users and there is no known explanation of it. Vast majority of users use it with 30s interval without problems, I do use 10s polling for years [...]

I'm polling every 500ms and have no issues with sems (GW10K-ET Plus).

smileyman-1, 2023-11-01:
I have a GW10KD-DT (SDT G2) that no longer sends data to SEMS below 180 seconds interval. Various EM series inverters, on the other hand, have no problem here.
 
The GW10KD-DT has the latest firmware (ARM and DSP) installed. The WiFi dongle also has the latest firmware (4.13.32). -> no problems with 1 second interval.
The EM series also has the latest firmware installed on the ARM / DSP and WiFi dongle.

Differences:
- The GW10KD-DT has a USB WiFi dongle installed, whereas the EM series has one with a COM port.

RobKuipers, 2024-03-06:
Same problem with a GW5000-SDT-20 (firmware V1.18.18.20) with Wifi-module (firmware version V1.1.3.10).

Any chance the new dongles ( [https://nl.goodwe.com/wifi-lan-20-and-wifi-20] ) could make any difference?

ivik, 2024-03-25:
> Same problem with a GW5000-SDT-20 (firmware V1.18.18.20) with Wifi-module (firmware version V1.1.3.10).
> 
> Any chance the new dongles ( [https://nl.goodwe.com/wifi-lan-20-and-wifi-20] ) could make any difference?

I have the new LAN/WIFI module, it does not work with HA or this Goodwe python library. I cannot connect whatever I do, but as soon as I change to the Wifi only module than everything works. But I also have issue with SEMS portal not getting any data as soon as I use HA.

mletenay, 2024-05-21:
Look at this comment https://github.com/home-assistant/core/issues/101817#issuecomment-2123445440 please and provide the diag infos, maybe? it would help ...

t-oster, 2024-09-30:
did anyone find a solution? I have the same problem. Some people suggest an interval of 127s, but for providing the value to my go-e controller this is too slow, I need at least every 30s, better every 10s.
Maybe the connection needs to be closed explicitely or something like that?

Misenkooo, 2025-03-10:
> I can report that the problem with this (at least for me ) was the fact I was using the goodwe wifi dongle 1.0 - Upgrading to the version 2.0 dongle resolved this matter and I now can query every second from two different systems and have the portal works fine.
> 
> This is a hardware issue plain and simple.

Could you please write down part number of this dongle ? I have Wifi dongle without LAN but not sure what version is it.