marcelblijleven/goodwe #51: PPV negative value on ET inverter

AdriVillaB, 2023-06-20:
Hi, 

I have a GW3600-EH inverter and some months ago I started to get a -2 W read on the ppv sensor when there is no solar production.

I was debugging the code and the problem seems to be on this line (https://github.com/marcelblijleven/goodwe/blob/29de1c4303fffd3e984eb3314db4b5085fbaf334/goodwe/et.py#L35). Although a filter to reject the ppv3 and ppv4 values is used, the ppv sensor gets populated using the Calculated class using a lambda function that pulls all the 4 current power.

The issue is that when accessing the ppv3 and ppv4 using the read_bytes4, the function returns a -1, therefore the sum will be 0+0+(-1)+(-1).

Could you help me to solve this issue? I don't know exactly how we could avoid reading this values?

Thanks!

oziee, 2023-06-27:
something like this would solve it
```
Calculated("ppv",
                   lambda data:
                   max(0, read_bytes4(data, 10)) +
                   max(0, read_bytes4(data, 18)) +
                   max(0, read_bytes4(data, 26)) +
                   max(0, read_bytes4(data, 34)),
                   "PV Power", "W", Kind.PV),
```

this just sets that is the value of the read_byte4 is a negative value set it to 0
so `0+0+(-1)+(-1)` would become `0+0+(0)+(0)`

jenicek001, 2023-06-27:
Hello,
this might be problem of inverter calibration from production - in my case GW10K-ET I have problem with offset between -0,3 and -0,2A on battery current measurement, so in cosequence it also incorrectly calculates battery energy charged. The same problem is visible in PVMaster / SolarGo. I solve it for more than 6 months with GoodWe support.
Regards,
Jan

AdriVillaB, 2023-06-28:
Hi @jenicek001,

I'm 100% sure the problem is not on the inverter as I was debugging the code. I'll try the solution @oziee suggest tonight and I'll update here.

Thanks

closed 2023-10-15 by mletenay