marcelblijleven/goodwe #53: no attribute connect

k4r3l01, 2023-07-22:
Hello please i have question im trying to use your code to get data from my GW10K-ET but when i try to run inverter_check.py and set ip address of my goodwe i get python error 
` inverter = asyncio.run(goodwe.connect(IP_ADDRESS, FAMILY, COMM_ADDR, TIMEOUT, RETRIES))
                           ^^^^^^^^^^^^^^
AttributeError: partially initialized module 'goodwe' has no attribute 'connect' (most likely due to a circular import)`

closed 2023-07-22 by k4r3l01

karolzarebski, 2024-02-17:
Hey, did you manage to solve this and?

zackdvd, 2024-08-28:
do not create a file goodwe.py, since it then has the same name as the module you want to import. 
Use any other name. 

can be closed now.