marcelblijleven/goodwe #55: Error fetching GoodWe data: ILLEGAL DATA ADDRESS

JoTec2002, 2023-08-08:
Hello,

I'm running an Goodwe gw15k-et Inverter connected via LAN.
Lan Module Firmware Version: V1.2.1.18
Inverter Firmware reportet in HA:  00300034003000360032002d / 02020-04-S01

I'm reciving an error in HA: Error fetching GoodWe data: ILLEGAL DATA ADDRESS

The Error is not present on every data fetch.

I alredy added
```
 logger:
  default: warning
  logs:
    custom_components.goodwe: debug
    goodwe: debug
```

but this didn't seem to get me more debug information.

mletenay, 2023-12-03:
Did you manage to get some actual logs ?

JoTec2002, 2023-12-04:
Sorry I forgot that I have this issue still open.
The Inverter was replaced since it didn't start up properly. The new one isn't installed yet.
I'm sorry for any inconvenience. I'm currently hoping that the issue will be resolved by the new inverter.