marcelblijleven/goodwe #57: Add new registers to ET

TomEkk33, 2023-10-28:
Hello,
Can you add 2 new registers to et.py file? As listed below:
[et_add_regs.patch](https://github.com/marcelblijleven/goodwe/files/13196638/et_add_regs.patch)
It is based on 0.2.32 version. I use this at my side with GW10k-ET inverter and it works as expected.

Thank you.

jenicek001, 2023-10-29:
Hello @TomEkk33, these new are RW registers controlled by SEC1000(S)?
These work only in General mode?
Thank you,
Jan

TomEkk33, 2023-10-29:
Hello @jenicek001,
These are GW10k-ET inverter only registers. They are not related to SEC1000(S).

jenicek001, 2023-10-29:
Hello @TomEkk33,

In standard operation the Inverter uses SmartMeter as a feedback - any load consuming energy (on any phase) pulls power from the grid, and Inverter compensates this power by generating the same amount of power. This runs in a closed loop as a regulator.

For use with SEC1000(S), there is no SmartMeter for each controlled inverter, SEC1000(S) has its own single SmartMeter and controls all connected inverters using their EMS ports to do the same, just it splits their usage to appropriate portion only.

This Python module behaves the same - it just does not use RS485 EMS port of Inverter, but uses the same via TCP or UDP.

What I am really curious is what is Inverter reaction to set_fixed_active_power() or set_fixed_reactive_power() - while connected with SmartMeter, Inverter uses its own regulation logic to generate active and reactive power (to specific phases). Do these functions actually work with SmartMeter connected? What is Inverter reaction to them actually?

I finally decided not to buy SEC1000S to my GW10K-ET and GW8K-ET inverters, which I am going to contol myself. However without SEC1000S I have to simply split loads between each single Inverter, because both have their own SmartMeter and all loads have to be connected in between, both Inverters are connected to grid behind their SmartMeters (they work independently).

Regards,
Jan

TomEkk33, 2023-10-30:
Hello,
> I use this at my side with GW10k-ET inverter and it works as expected.

I meant that values can be read/written. Nothing else was checked.

jenicek001, 2023-10-31:
Hello, do you have SmartMeter connected to Inverter or not?
What does the Inverter do after receiving the command set_fixed_active_power() or set_fixed_reactive_power()?
Thanks,
Jan

mletenay, 2023-12-21:
I do not think we should add those registers to library and general use.
They are potentially dangerous and should not be exposed to general public.
If you are expert user and know what you are doing, you can achieve that by using the low level API.

closed 2023-12-21 by mletenay

TomEkk33, 2023-12-21:
I'd like to see regs in library ONLY. Is it possible? It should not be added to higher level.