marcelblijleven/goodwe #62: GW20K-ET Wrong MPPT Power register mappings

AndreiLux, 2023-11-19:
Hi, 

Just got my 25K-ET installed a few days ago and still experimenting with everything.

Noticed there's a large discrepancy in the power reported by SEMS/SolarGo and also the ModBus outputs.

When reading out the registers via modbus as configured right now, you get the following:

```
vpv1: 				 PV1 Voltage = 696.9 V
ipv1: 				 PV1 Current = 3.3 A
ppv1: 				 PV1 Power = 1729 W
vpv2: 				 PV2 Voltage = 696.9 V
ipv2: 				 PV2 Current = 0.0 A
ppv2: 				 PV2 Power = 2922 W
vpv3: 				 PV3 Voltage = 647.4 V
ipv3: 				 PV3 Current = 5.3 A
ppv3: 				 PV3 Power = 0 W
ppv: 				 PV Power = 4651 W
```

For context, my configuration population is 1 string each on both MPPT1 and MPPT2, with MPPT3 unpopulated.

The problem as you see in the above breakdown, is that the power calculation as described by the power registers isn't accurate.

In comparison, this is what SolarGo is reporting, which seems to be accurate as far as the population config:

[image]

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Andrei/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Andrei/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  | V | A | P calculated | P modbus
-- | -- | -- | -- | --
PV1 | 696 | 3.3 | 2296.8 | 1729
PV2 | 696 | 0 | 0 | 2922
PV3 | 647 | 5.3 | 3429.1 | 0
PV4 | 647 | 0 | 0 | n/a
PV5 | 0 | 0 | 0 | n/a
PV6 | 0 | 0 | 0 | n/a
  |   | Total P | 5725.9 | 4651

</body>

</html>

In the SolarGo flow overview, the app is showcasing the 5.7kW figure as generation coming from the panels.

When I go into SEMS, that is reporting the 4.6kW figure, which seems to match the modbus power total.

What's actually even more odd, is that although the calculated figure _seems_ more accurate, the breakdown of where the power is flowing afterwards seems to want to indicate that the ModBus/SEMS figure to be the accurate ones, as it's showing ~2.3kW into the backup/grid output and 2.3kW into the battery and 0W into the grid meter.

However, SolarGo is showing 3.5kW into the house, which I _know_ to be inaccurate, and I also noted that SolarGo always overestimates the house power by a large amount which seems to correlate with the calculated vs mdobus power sum difference.

To sum it up, it's one giant mess across all the apps and reporting figures.

At the very least, what is certain, is that the modbus ppv2/ppv3 figures are not showcasing what they're labelled as, assuming the values they are showing are even accurate. Instead what they seem to be showing is per-MPPT string-aggregate power figures, and the V/A registers are showing per-string data, hence the mismatch in the labelling.

Unfortunately I don't have MPPT3 populated as it would be interesting to see what the modbus output even shows when power would be flowing through that, given that PV4/PV5/PV6 data doesn't exist in the current known mappings (I saw the mention that one might attempt reverse-engineer things from the app).

Another FYI: I have the whole house exclusively connected to the backup output, however one will note that the modbus pgrid/2/3 figures will be +- mirroring the backup load outputs even though they should be zero (as shown by the meter itself), so that data is also relatively untrustworthy and useless and forces you to refer to the meter data and calculations to do any possible grid-tied loads.

AndreiLux, 2023-11-19:
Here's also a bit more context as to what those "pgrid" registers are:

`C:\Users\Andrei\AppData\Local\Programs\Python\Python311\python.exe F:\Workfolder\GoodWe\main.py 
timestamp: 				 Timestamp = 2023-11-19 13:55:20 
vpv1: 				 PV1 Voltage = 690.9 V
ipv1: 				 PV1 Current = 2.5 A
ppv1: 				 PV1 Power = 1188 W
vpv2: 				 PV2 Voltage = 690.9 V
ipv2: 				 PV2 Current = 0.0 A
ppv2: 				 PV2 Power = 719 W
vpv3: 				 PV3 Voltage = 725.2 V
ipv3: 				 PV3 Current = 1.9 A
ppv3: 				 PV3 Power = 0 W
ppv: 				 PV Power = 1907 W
pv3_mode: 				 PV3 Mode code = 0 
pv3_mode_label: 				 PV3 Mode = PV panels not connected 
pv2_mode: 				 PV2 Mode code = 2 
pv2_mode_label: 				 PV2 Mode = PV panels connected, producing power 
pv1_mode: 				 PV1 Mode code = 2 
pv1_mode_label: 				 PV1 Mode = PV panels connected, producing power 
vgrid: 				 On-grid L1 Voltage = 231.8 V
igrid: 				 On-grid L1 Current = 0.7 A
fgrid: 				 On-grid L1 Frequency = 49.98 Hz
pgrid: 				 On-grid L1 Power = -9 W
vgrid2: 				 On-grid L2 Voltage = 232.5 V
igrid2: 				 On-grid L2 Current = 10.2 A
fgrid2: 				 On-grid L2 Frequency = 49.98 Hz
pgrid2: 				 On-grid L2 Power = 1943 W
vgrid3: 				 On-grid L3 Voltage = 231.8 V
igrid3: 				 On-grid L3 Current = 2.7 A
fgrid3: 				 On-grid L3 Frequency = 49.98 Hz
pgrid3: 				 On-grid L3 Power = 528 W
grid_mode: 				 Grid Mode code = 1 
grid_mode_label: 				 Grid Mode = Connected to grid 
total_inverter_power: 				 Total Power = 2462 W
active_power: 				 Active Power = -27 W
grid_in_out: 				 On-grid Mode code = 0 
grid_in_out_label: 				 On-grid Mode = Idle 
reactive_power: 				 Reactive Power = 1804 var
apparent_power: 				 Apparent Power = 3158 VA
backup_v1: 				 Back-up L1 Voltage = 230.4 V
backup_i1: 				 Back-up L1 Current = 3.0 A
backup_f1: 				 Back-up L1 Frequency = 49.98 Hz
load_mode1: 				 Load Mode L1 = 0 
backup_p1: 				 Back-up L1 Power = 49 W
backup_v2: 				 Back-up L2 Voltage = 231.9 V
backup_i2: 				 Back-up L2 Current = 8.8 A
backup_f2: 				 Back-up L2 Frequency = 49.98 Hz
load_mode2: 				 Load Mode L2 = 0 
backup_p2: 				 Back-up L2 Power = 1986 W
backup_v3: 				 Back-up L3 Voltage = 230.8 V
backup_i3: 				 Back-up L3 Current = 3.2 A
backup_f3: 				 Back-up L3 Frequency = 49.98 Hz
load_mode3: 				 Load Mode L3 = 0 
backup_p3: 				 Back-up L3 Power = 363 W
load_p1: 				 Load L1 = 46 W
load_p2: 				 Load L2 = 2000 W
load_p3: 				 Load L3 = 360 W
backup_ptotal: 				 Back-up Load = 2386 W
load_ptotal: 				 Load = 103 W
ups_load: 				 Ups Load = 14 %
temperature_air: 				 Inverter Temperature (Air) = 43.5 C
temperature_module: 				 Inverter Temperature (Module) = 0.0 C
temperature: 				 Inverter Temperature (Radiator) = 39.6 C
function_bit: 				 Function Bit = 0 
bus_voltage: 				 Bus Voltage = 756.7 V
nbus_voltage: 				 NBus Voltage = 378.3 V
vbattery1: 				 Battery Voltage = 212.5 V
ibattery1: 				 Battery Current = 1.9 A
pbattery1: 				 Battery Power = 404 W
[... 70 lines cut ...]
meter_active_power2: 				 Meter Active Power L2 = -43 W
meter_active_power3: 				 Meter Active Power L3 = 193 W
meter_active_power_total: 				 Meter Active Power Total = 116 W
meter_reactive_power1: 				 Meter Reactive Power L1 = 413 var
meter_reactive_power2: 				 Meter Reactive Power L2 = -1025 var
meter_reactive_power3: 				 Meter Reactive Power L2 = 213 var
meter_reactive_power_total: 				 Meter Reactive Power Total = -398 var
meter_apparent_power1: 				 Meter Apparent Power L1 = -444 VA
meter_apparent_power2: 				 Meter Apparent Power L2 = -1099 VA
meter_apparent_power3: 				 Meter Apparent Power L3 = 368 VA
meter_apparent_power_total: 				 Meter Apparent Power Total = 1912 VA
meter_type: 				 Meter Type = 2 
meter_sw_version: 				 Meter Software Version = 5 `

On these ETT series, they are **not** on-grid outputs but rather they are the sum of the AC output generated by the inverter through PV and battery, wherever they might go. SolarGo is also wrongly labelling these metrics in the app as "on-grid" which can be a bit confusing. I also remember that when these values go negative is when the inverter has to ingest power from the grid connectors, but I need to check again during the night and when the battery is discharged.

In the above snippet, PV is generating 1907W (ppv), the battery is generating 404W (pbattery1), the inverter total AC generation is 2462W (total_inverter_power ~= sum of pgrid/2/3), with a consumption hanging off the backup load of 2386W (backup_ptotal). In this scenario load_p1/2/3 ~= backup_p1/2/3. It's ingesting 27W from the grid side connectors (active_power).

active_power +- equals the meter_active_power_total in my scenario as I have basically nothing connected to non-backup except the distribution panel itself and whatever losses might happen in there between the CTs and the actual inverter.

active_power goes positive when injecting into the grid connector, and goes negative when ingesting through it.

Also with the above behaviour, "house_consumption" calculation as it is currently is wrong and off by a bit as it doesn't end up matching the actual load, a simpler way to generate a more accurate figure is to just do "total_inverter_power - active_power".

I'll add more comments once I manage to correlate the numbers in different scenarios.

AndreiLux, 2023-11-25:
> However, SolarGo is showing 3.5kW into the house, which I know to be inaccurate, and I also noted that SolarGo always overestimates the house power by a large amount which seems to correlate with the calculated vs mdobus power sum difference.

Update: This behaviour has now been fixed as of DSP V06 SVN 6017 on ETT models, and now the calculated reported modbus V/A on the strings matches the MPPT power entries. The labelling mismatch still exists (V/A entries == string #'s, P entries == MPPT #'s).

AndreiLux, 2023-11-25:
Please find attached a complete revamped modbus register mapping for the ES G2 and ETT/ET-15-30 inverters.

Somebody simply just had to ask GoodWe for it and we got it.

[ARM 745, ESG2,ET30 MODBUS Protocol Map 20221231 - v1.pdf](https://github.com/marcelblijleven/goodwe/files/13464476/ARM.745.ESG2.ET30.MODBUS.Protocol.Map.20221231.-.v1.pdf)

Credit to "NorthPV": https://www.photovoltaikforum.com/thread/186135-goodwe-gw10k-et-sammelthread/?postID=3488118#post3488118

Due the extensive differences here, it makes sense to split off those inverter series into a different mapping file and not attempt to include it into the current et.py. I will look into it as I have spare time, but hopefully this allows the community to advance at greater pace.

mletenay, 2023-11-27:
That's great. AFAIK Goodwe was not really willing to share such data without NDA in the past, so this is definitely progress.
I will study it and incorporate it in the code sometime this week ...

mletenay, 2023-12-03:
The code was enhanced with ETT specific registers besides support for 2nd battery it now provides values for PV5-16 strings and power and current values for MPPT 1-8.