marcelblijleven/goodwe #69: GW25K-ET missing voltage & ampere of 2nd battery

MirrorLink700, 2024-01-21:
Hi, thanks again for adding the soc of the 2nd battery. The output for the 2nd battery doesn't contain the voltage and ampere, so the power use of the 2nd battery can't be calculated. Or is there an other way to calculate the pbattery2 value out of house_consumption value?

mletenay, 2024-01-29:
Hmm, it seems inverter does not report 2nd battery voltage/current, there is no known register with those values.
And the house_consumption is calculated value (from pbattery1 register) ...
I'm afraid there's no help at the moment.

MirrorLink700, 2024-02-29:
Register 35264 (signed32) is power of 2nd Battery. 47924(U16) is voltage and 47925(U16) is current. Could you please verify and implement this?

mletenay, 2024-02-29:
Hmm, where did you get the 35264 register description ? I do not have it in any of public/not-so-public sources

MirrorLink700, 2024-03-01:
There is a discussion ongoing on photovoltaikforum 
https://www.photovoltaikforum.com/thread/204490-goodwe-gw25k-et-sammelthread/?pageNo=37

pravussum, 2025-02-23:
The missing power value for the 2nd battery nags me for quite some time. Un-commenting the line [here](https://github.com/marcelblijleven/goodwe/blob/master/goodwe/et.py#L156) just leads to 0 being returned all the time (even though the 2nd battery is (dis)charging. 

With a little debugging I found that the value from the buffer read [here](https://github.com/marcelblijleven/goodwe/blob/master/goodwe/sensor.py#L857) is empty (not 0). Didn't found out why yet. 

Besides that, the SolarGo App shows the battery 2 power, so the value must be there. I sniffed the traffic from Solargo to the inverter and this is what I got:

Beside the request to register 0x891c (count 125 = 250 bytes) it makes another request to 0x89be with count 6 (12 bytes).
The bytes marked in the screenshot are 4 bytes starting of address 0x89c0 = 35264, the address of the battery 2 power. So the value is definetely returned by the inverter.

[image]

`ff ff fb 1a` interpreted as a signed int is 1253, which is exactly what Solargo showed in the GUI for battery 2 (1,25 kW).

pravussum, 2025-02-23:
The offset of 35264 to the beginning of runtime data (35100) is 164 registers, but only 125 are read. So the data is obviously not part of the buffer (took a while to sink in). 
Just reading a larger block ([here](https://github.com/marcelblijleven/goodwe/blob/master/goodwe/et.py#L475)) didn't work, so I think a new block needs to be introduced.