marcelblijleven/goodwe #74: newest version in Code no longer works with my GW3600-EH

janjos0, 2024-04-26:
Hi, 
Just to inform that using release version 0.3.3, I can get/read the data from my GW3600-EH.
With the newest versions in Code
https://github.com/marcelblijleven/goodwe/tree/master/goodwe
I always get error messages
Unable to connect to the inverter at ...
No valid response received ...

Maybe it is just my limited knowledge how to do it. I use your code in a Windows 10 machine where I made a folder that includes the goodwe folder with the files from the library and a .py file with your example code to read the data:

import asyncio
import goodwe

async def get_runtime_data():
    ip_address = '192.168.1.14'

    inverter = await goodwe.connect(ip_address)
    runtime_data = await inverter.read_runtime_data()

    for sensor in inverter.sensors():
        if sensor.id_ in runtime_data:
            print(f"{sensor.id_}: \t\t {sensor.name} = {runtime_data[sensor.id_]} {sensor.unit}")

asyncio.run(get_runtime_data())

mletenay, 2024-04-26:
Thanks for feedback !
The redesign I did due to the new TCP stuff was rather big and I might broke something.
It worked form my IDE, but I need to re-test it.

mletenay, 2024-04-28:
I indeed forgot to add new port parameter to the initial discovery method !
It should be fixed now.
Thanks for testing !

closed 2024-04-28 by mletenay

janjos0, 2024-04-29:
You are the one we must thank, so thank you for this code.

Unfortunately still does not work with my GW3600-EH.

This is what I get when using the newest files from https://github.com/marcelblijleven/goodwe/tree/master/goodwe :
Traceback (most recent call last):
  File "c:\users\public\software\python\code\mygoodwev2\goodwe\inverter.py", line 143, in _read_from_socket
    result = await command.execute(self._protocol)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\public\software\python\code\mygoodwev2\goodwe\protocol.py", line 319, in execute
    response_future = await protocol.send_request(self)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\public\software\python\code\mygoodwev2\goodwe\protocol.py", line 106, in send_request
    await response_future
goodwe.exceptions.MaxRetriesException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\public\software\python\code\mygoodwev2\mygoodwev2.py", line 16, in <module>
    asyncio.run(get_runtime_data())
  File "C:\Program Files\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "c:\users\public\software\python\code\mygoodwev2\mygoodwev2.py", line 9, in get_runtime_data
    runtime_data = await inverter.read_runtime_data()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\public\software\python\code\mygoodwev2\goodwe\et.py", line 502, in read_runtime_data
    response = await self._read_from_socket(self._READ_RUNNING_DATA)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\public\software\python\code\mygoodwev2\goodwe\inverter.py", line 148, in _read_from_socket
    raise RequestFailedException(f'No valid response received even after {self._protocol.retries} retries',
goodwe.exceptions.RequestFailedException: ('No valid response received even after 3 retries', 1)

And this is what I get when using the files from the published version 0.3.3:

timestamp:               Timestamp = 2024-04-29 08:52:48
vpv1:            PV1 Voltage = 261.4 V
ipv1:            PV1 Current = 0.5 A
ppv1:            PV1 Power = 130 W
vpv2:            PV2 Voltage = 117.4 V
ipv2:            PV2 Current = 1.1 A
ppv2:            PV2 Power = 129 W
ppv:             PV Power = 259 W
pv2_mode:                PV2 Mode code = 2
pv2_mode_label:                  PV2 Mode = PV panels connected, producing power
pv1_mode:                PV1 Mode code = 2
pv1_mode_label:                  PV1 Mode = PV panels connected, producing power
vgrid:           On-grid L1 Voltage = 234.8 V
igrid:           On-grid L1 Current = 1.1 A
fgrid:           On-grid L1 Frequency = 50.03 Hz
pgrid:           On-grid L1 Power = 261 W
grid_mode:               Grid Mode code = 1
grid_mode_label:                 Grid Mode = Connected to grid
total_inverter_power:            Total Power = 261 W
active_power:            Active Power = 261 W
grid_in_out:             On-grid Mode code = 1
grid_in_out_label:               On-grid Mode = Exporting
reactive_power:                  Reactive Power = -1 var
apparent_power:                  Apparent Power = -1 VA
backup_v1:               Back-up L1 Voltage = 0.0 V
backup_i1:               Back-up L1 Current = 0.0 A
backup_f1:               Back-up L1 Frequency = 0.0 Hz
load_mode1:              Load Mode L1 = 0
[... 38 lines cut ...]
diagnose_result_label:           Diag Status = Battery voltage low, Battery SOC low, Battery SOC in back, BMS: Discharge disabled, Discharge Driver On, BMS: Discharge current low, Meter communication failure, Battery Disconnected, Self-use off, Export power limit set, PF value set
house_consumption:               House Consumption = -2 W
commode:                 Commode = 1
rssi:            RSSI = 101
manufacture_code:                Manufacture Code = 10
meter_test_status:               Meter Test Status = 0
meter_comm_status:               Meter Communication Status = 0
active_power1:           Active Power L1 = 0 W
active_power_total:              Active Power Total = 0 W
reactive_power_total:            Reactive Power Total = 0 var
meter_power_factor1:             Meter Power Factor L1 = 0.0
meter_power_factor:              Meter Power Factor = 0.0
meter_freq:              Meter Frequency = 0.0 Hz
meter_e_total_exp:               Meter Total Energy (export) = 0.0 kWh
meter_e_total_imp:               Meter Total Energy (import) = 0.0 kWh
meter_active_power1:             Meter Active Power L1 = 0 W
meter_active_power_total:                Meter Active Power Total = 0 W
meter_reactive_power1:           Meter Reactive Power L1 = 0 var
meter_reactive_power_total:              Meter Reactive Power Total = 0 var
meter_apparent_power1:           Meter Apparent Power L1 = 0 VA
meter_apparent_power_total:              Meter Apparent Power Total = 0 VA
meter_type:              Meter Type = 0
meter_sw_version:                Meter Software Version = 0

I connect to the inverter via its own WiFi on IP 10.10.100.253 and, I think, using UDP protocol.

mletenay, 2024-04-29:
Is it 100% reproducible ?
Since the UDP (and inverter wifi module) is by definition unreliable and it sometimes happen it does not reply in time even after 3 retries (the retry and timeout are configurable).
That exception is raised when it failed after 3 attempts.
Please try again couple of times and let me know ...

janjos0, 2024-04-30:
Yes, it is reproducible.
I tested yesterday several times and also this morning, >10.
I don't know what is the time between retries and timeout period, but with the newest files it shows the errors like just two seconds after starting the program from a Windows cmd line.
If I use the 0.3.3 version it works fine.

Can you please confirm that the example code in the README -> USAGE and after writing the correct IP, 10.10.100.253 in my case, is still valid to test the new files ?
I ask this, because like I said before, my programming knowledge is very limited.

mletenay, 2024-04-30:
You are using it the correct way, I just introduced there couple of bugs ;-(
It should be fixed now, please re-test/

janjos0, 2024-05-01:
Now it works.
I do have the feeling that the connection fails more often but it surely works.
Thank you.

Something else about the data from my inverter, I don't know if you noticed this (wrong) reading
temperature_module: Inverter Temperature (Module) = 3276.7 C
It always shows this very high temperature value.
Maybe it is because my type of inverter does not have this sensor built-in, so the register just reports erroneous data.

mletenay, 2024-05-03:
Yes, it seem EH inverter do not support module temperature sensors and return 0x7FFF.
I've just fixed that so it is reported as None.