marcelblijleven/goodwe #77: Modbus registers issues/adds on GW10k-ET on fw:18+

frantisadlo, 2024-05-04:
Hi Marcel,

  I am really sorry for inconviniences from this form, but I am not so sure how to manage PULLs in Git correctly and do not want to make any mess.

I am using this library standalone without HA and I identified some issuses which is not visible with HA, maybe.

I have been test it on few different GW10k-ET inverters in production with fw23 and fw111127 (identified by SolarGo).

Main issue, which is raising exception on Invertor.read_settings_data() class function, because some registers are not accesible (trace is below). It is all those 3:  
`Integer("max_charge_power", 47606, "Max Charge Power"),
 Integer("smart_charging_enable", 47609, "Smart Charging Mode Switch"),
Integer("eco_mode_enable", 47612, "Eco Mode Switch"),`

failing with ILLEGAL DATA ADDRESS response. I do not want to change your code so much, then I do not go too deep to solve handling missing/wrong registers in general in this function, but it can be helpfull to implement it with as just log warnning and not to lead Exit or just hint me how to handle it correctly, for now RequestRejectedException is not propagated to call this function.

`Traceback (most recent call last):
  File "/home/epm/program/epm_ec/devices/epm_goodwe.py", line 119, in <module>
    asyncio.run(check(inv_dev))
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/devices/epm_goodwe.py", line 108, in check
    st_data = await inv.read_settings_data()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/devices/epm_goodwe.py", line 84, in read_settings_data
    value = await self.dev.read_setting(setting.id_)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/et.py", line 611, in read_setting
    return await self._read_setting(setting)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/et.py", line 615, in _read_setting
    response = await self._read_from_socket(self._read_command(setting.offset, count))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/inverter.py", line 124, in _read_from_socket
    result = await command.execute(self._protocol)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/protocol.py", line 372, in execute
    response_future = await protocol.send_request(self)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/protocol.py", line 135, in send_request
    await response_future
  File "/home/epm/program/epm_ec/goodwe/protocol.py", line 111, in datagram_received
    if self.command.validator(data):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/protocol.py", line 501, in <lambda>
    lambda x: validate_modbus_rtu_response(x, cmd, offset, value),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/epm/program/epm_ec/goodwe/modbus.py", line 205, in validate_modbus_rtu_response
    raise RequestRejectedException(failure_code)
goodwe.exceptions.RequestRejectedException: ILLEGAL DATA ADDRESS`

In addition I have found from different sources (mainly in GW modbus doc v 1.10) and then tested on same inverters some extra registers which is not included in library and can be usefull. 
I recognize som changes in numbering (shifts +1), but testing show me that it is not correct, then I am not implement it).

My add-on is done in this way:

`# Settings added in ARM firmware 18
    __settings_arm_fw_18: Tuple[Sensor, ...] = (
        # Load regulation 0: Disable, 1: Switching mode, 2: Time manage mode
        Integer("load_regulation_index", 47595, "Load Regulation Index"),
        # Load switch status: 0/1
        Integer("load_switch_status", 47596, "Load Switch status"),
        # Backup Switch SOC Min: 0/100 in percent
        Integer("backup_switch_soc_min", 47597, "Backup Switch SOC Min."),
        # Hardware Feed Power> 0/1
        Integer("hardware_feed_power", 47599, "Hardware Feed Power"),
        # Direct BMS communication for EMS Control
        Integer("bms_version", 47900, "BMS Version"),
        Integer("batt_strings", 47901, "Battery Strings"),
        # Real time read from BMS
        Voltage("bms_bat_charge_v_max", 47902, "BMS max. Charge Voltage limit", Kind.BMS),
        Current("bms_bat_charge_i_max", 47903, "BMS max. Charge Current limit", Kind.BMS),
        Voltage("bms_bat_discharge_v_min", 47904, "BMS min. Discharge Voltage limit", Kind.BMS),
        Current("bms_bat_discharge_i_max", 47905, "BMS max. Discharge Current limit", Kind.BMS),
        Voltage("bms_bat_voltage", 47906, "BMS Battery Voltage", Kind.BMS),
        Current("bms_bat_current", 47907, "BMS Battery Current", Kind.BMS),
        #
        Integer("bms_bat_soc", 47908, "BMS Battery SOC", "%", Kind.BMS),
        Integer("bms_bat_soh", 47909, "BMS Battery SOH", "%", Kind.BMS),
        Temp("bms_bat_temperature", 47910, "BMS Battery Temperature", Kind.BMS),
        Long("bms_bat_warning-code", 47911, "BMS Battery Warning Code"),
        # Reserved
        Long("bms_bat_alarm-code", 47913, "BMS Battery Alarm Code"),
        Integer("bms_status", 47915, "BMS Status"),
        Integer("bms_comm_loss_disable", 47916, "BMS Communication Loss Disable"),
        # RW settings of BMS voltage rate
        Integer("bms_battery_string_rate_v", 47917, "BMS Battery String Rate Voltage"),
    )
`

Plus device_info has some minor issue. I add "ID of GPRS/4g module" and commented some FW ids which is not documented in Modbuse protocol sheet 1.10 from 25. 3. 2022:
`      async def read_device_info(self):
        response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
        response = response.response_data()
        # Modbus registers from offset (35000)
        self.modbus_version = read_unsigned_int(response, 0)
        self.rated_power = read_unsigned_int(response, 2)
        self.ac_output_type = read_unsigned_int(response, 4)  # 0: 1-phase, 1: 3-phase (4 wire), 2: 3-phase (3 wire)
        self.serial_number = self._decode(response[6:22])
        self.model_name = self._decode(response[22:32])
        self.dsp1_version = read_unsigned_int(response, 32)
        self.dsp2_version = read_unsigned_int(response, 34)
        self.dsp_svn_version = read_unsigned_int(response, 36)
        self.arm_version = read_unsigned_int(response, 38)
        self.arm_svn_version = read_unsigned_int(response, 40)
        self.sim_cc_id = self._decode(response[50:60])
        # not documented
        # self.firmware = self._decode(response[42:54])
        # self.arm_firmware = self._decode(response[54:66])`

`class Inverter(ABC):
    """
    Common superclass for various inverter models implementations.
    Represents the inverter state and its basic behavior
    """
    def __init__(self, host: str, port: int, comm_addr: int = 0, timeout: int = 1, retries: int = 3):
        self._protocol: InverterProtocol = self._create_protocol(host, port, timeout, retries)
        self._consecutive_failures_count: int = 0
        self.comm_addr: int = comm_addr
        self.model_name: str | None = None
        self.serial_number: str | None = None
        self.rated_power: int = 0
        self.ac_output_type: int | None = None
        self.firmware: str | None = None
        self.arm_firmware: str | None = None
        self.modbus_version: int | None = None
        self.dsp1_version: int = 0
        self.dsp2_version: int = 0
        self.dsp_svn_version: int | None = None
        self.arm_version: int = 0
        self.arm_svn_version: int | None = None
        self.sim_cc_id: str | None = None`

So I am able to provide my current files which I updated. ET.py and Inverter.py with some hints how to do that correctly or just simply send them if it is makes sense to you.

Thank you for your gret big work on this library. It is really usefull and I hope that I help you a bit.
BR
Frantisek

mletenay, 2024-05-05:
Thanks for feedback, but it is really difficult to process it in this form.
You don't have to create a PR, just fork and clone this repository and commit changes to your clone.
I can can take a look at those commits and adapt this master accordingly.

To your suggestions - bottom up.

1) The sim_cc_id is not correct. 
It cannot be decoded from response[50:60] - those are modbus registers 35025-35029 and they obviously overlap with the firmware/arm_firmware strings. Which are not present in documents but are present there in real life (and proved by tests).
The docs says the SIMCCID - ID of GPRS/4G module is at modbus #35050 which would have been on offset 35100, but that one is outside range read by _READ_DEVICE_VERSION_INFO command.

2) Not all registers are present, only the most useful. 
Some may not have been known at the time, some may be considered useless and some right away dangerous.
If some very advanced user wants to access those registers, there's always the possibility to invoke explicit low-level modbus read command - see inverter_check.py around line 90, 

3) The ILLEGAL DATA ADDRESS is weird and we should investigate it in detail.
I have fw23 for quite some time and there are no problems.
Which specific firmwares do fail ?

frantisadlo, 2024-05-05:
Hi Marc, look like I made a fork. Hope it works as you expected.

ad 1) You are right, it is my mystyping/misunderstanding of code (forgot multiply 2). No valid return.

ad 2) Sure. No ofense. Just because I am looking for check direct BMS data to validate correct settings from installers guys, I offer it, just if somebody can use this as well. Data from 45352 are in Inverter side (by their DOC), then I had some mismatch issues to solve.

ad3) Here is my full output from inverter with 23 version after SIM corrections see ad1) and slight anonymizations. Does it help or how can I help you more?

sudo+ssh://epm@192.168.x.y:22/home/epm/program/epm_ec/.venv/bin/python -u /home/epm/program/epm_ec/devices/epm_goodwe.py
INFO:__main__:Inverter identified as GW10K-ET SN:9010KETU218Wxxxx.
Created inverter on IP: 192.168.x.y has following attributes:
_protocol:  <goodwe.protocol.UdpInverterProtocol object at 0x7fabeed090>
_consecutive_failures_count:  0
comm_addr:  247
model_name:  GW10K-ET
serial_number:  9010KETU218Wxxxx
rated_power:  10000
ac_output_type:  254
firmware:  04029-09-S11
arm_firmware:  02041-23-S00
modbus_version:  2
dsp1_version:  9
dsp2_version:  9
dsp_svn_version:  161
arm_version:  23
arm_svn_version:  241
_READ_DEVICE_VERSION_INFO:  READ 33 registers from 35000 (f70388b800213ac1)
_READ_RUNNING_DATA:  READ 125 registers from 35100 (f703891c007d7ae7)
_READ_METER_DATA:  READ 45 registers from 36000 (f7038ca0002dbbf3)
_READ_METER_DATA_EXTENDED:  READ 58 registers from 36000 (f7038ca0003afbfd)
_READ_BATTERY_INFO:  READ 24 registers from 37000 (f70390880018fc7c)
_READ_BATTERY2_INFO:  READ 22 registers from 39000 (f703985800167e21)
_READ_MPPT_DATA:  READ 61 registers from 35301 (f70389e5003dab26)
_has_eco_mode_v2:  True
_has_peak_shaving:  True
_has_battery:  True
_has_battery2:  False
_has_meter_extended:  False
_has_mppt:  False
_sensors:  (Timestamp(id_='timestamp', offset=35100, name='Timestamp', size_=6, unit='', kind=None), Voltage(id_='vpv1', offset=35103, name='PV1 Voltage', size_=2, unit='V', kind=<SensorKind.PV: 1>), Current(id_='ipv1', offset=35104, name='PV1 Current', size_=2, unit='A', kind=<SensorKind.PV: 1>), Power4(id_='ppv1', offset=35105, name='PV1 Power', size_=4, unit='W', kind=<SensorKind.PV: 1>), Voltage(id_='vpv2', offset=35107, name='PV2 Voltage', size_=2, unit='V', kind=<SensorKind.PV: 1>), Current(id_='ipv2', offset=35108, name='PV2 Current', size_=2, unit='A', kind=<SensorKind.PV: 1>), Power4(id_='ppv2', offset=35109, name='PV2 Power', size_=4, unit='W', kind=<SensorKind.PV: 1>), Calculated(id_='ppv', offset=0, name='PV Power', size_=0, unit='W', kind=<SensorKind.PV: 1>), ByteH(id_='pv2_mode', offset=35120, name='PV2 Mode code', size_=1, unit='', kind=<SensorKind.PV: 1>), EnumH(id_='pv2_mode_label', offset=35120, name='PV2 Mode', size_=1, unit='', kind=<SensorKind.PV: 1>), ByteL(id_='pv1_mode', offset=35120, name='PV1 Mode code', size_=1, unit='', kind=<SensorKind.PV: 1>), EnumL(id_='pv1_mode_label', offset=35120, name='PV1 Mode', size_=1, unit='', kind=<SensorKind.PV: 1>), Voltage(id_='vgrid', offset=35121, name='On-grid L1 Voltage', size_=2, unit='V', kind=<SensorKind.AC: 2>), Current(id_='igrid', offset=35122, name='On-grid L1 Current', size_=2, unit='A', kind=<SensorKind.AC: 2>), Frequency(id_='fgrid', offset=35123, name='On-grid L1 Frequency', size_=2, unit='Hz', kind=<SensorKind.AC: 2>), PowerS(id_='pgrid', offset=35125, name='On-grid L1 Power', size_=2, unit='W', kind=<SensorKind.AC: 2>), Voltage(id_='vgrid2', offset=35126, name='On-grid L2 Voltage', size_=2, unit='V', kind=<SensorKind.AC: 2>), Current(id_='igrid2', offset=35127, name='On-grid L2 Current', size_=2, unit='A', kind=<SensorKind.AC: 2>), Frequency(id_='fgrid2', offset=35128, name='On-grid L2 Frequency', size_=2, unit='Hz', kind=<SensorKind.AC: 2>), PowerS(id_='pgrid2', offset=35130, name='On-grid L2 Power', size_=2, unit='W', kind=<SensorKind.AC: 2>), Voltage(id_='vgrid3', offset=35131, name='On-grid L3 Voltage', size_=2, unit='V', kind=<SensorKind.AC: 2>), Current(id_='igrid3', offset=35132, name='On-grid L3 Current', size_=2, unit='A', kind=<SensorKind.AC: 2>), Frequency(id_='fgrid3', offset=35133, name='On-grid L3 Frequency', size_=2, unit='Hz', kind=<SensorKind.AC: 2>), PowerS(id_='pgrid3', offset=35135, name='On-grid L3 Power', size_=2, unit='W', kind=<SensorKind.AC: 2>), Integer(id_='grid_mode', offset=35136, name='Grid Mode code', size_=2, unit='', kind=<SensorKind.PV: 1>), Enum2(id_='grid_mode_label', offset=35136, name='Grid Mode', size_=2, unit='', kind=<SensorKind.PV: 1>), PowerS(id_='total_inverter_power', offset=35138, name='Total Power', size_=2, unit='W', kind=<SensorKind.AC: 2>), PowerS(id_='active_power', offset=35140, name='Active Power', size_=2, unit='W', kind=<SensorKind.GRID: 5>), Calculated(id_='grid_in_out', offset=0, name='On-grid Mode code', size_=0, unit='', kind=<SensorKind.GRID: 5>), EnumCalculated(id_='grid_in_out_label', offset=0, name='On-grid Mode', size_=0, unit='', kind=<SensorKind.GRID: 5>), Reactive(id_='reactive_power', offset=35142, name='Reactive Power', size_=2, unit='var', kind=<SensorKind.GRID: 5>), Apparent(id_='apparent_power', offset=35144, name='Apparent Power', size_=2, unit='VA', kind=<SensorKind.GRID: 5>), Voltage(id_='backup_v1', offset=35145, name='Back-up L1 Voltage', size_=2, unit='V', kind=<SensorKind.UPS: 3>), Current(id_='backup_i1', offset=35146, name='Back-up L1 Current', size_=2, unit='A', kind=<SensorKind.UPS: 3>), Frequency(id_='backup_f1 [... cut ...]
fast_charging_soc: 		 Fast Charging SoC = 0 %
load_control_mode: 		 Load Control Mode = 0 
load_control_switch: 		 Load Control Switch = 0 
load_control_soc: 		 Load Control SoC = 0 
fast_charging_power: 		 Fast Charging Power = 100 %
peak_shaving_power_limit: 		 Peak Shaving Power Limit = 0 
peak_shaving_soc: 		 Peak Shaving SoC = 0 
peak_shaving_mode: 		 Peak Shaving Mode = 48:0-48:0  100% (SoC 100%) Unset 
dod_holding: 		 DoD Holding = 0 
backup_mode_enable: 		 Backup Mode Switch = 0 
Read. Sensors: 145 Settings: 63

mletenay, 2024-05-05:
Ad 2) - I am not saying we should not any new settings if we identify some useful and safe. 
I was just explaining why not everything is there.
Looking at the registers 45352-45358, yes they can make sense and we can add them.

I do not see any commit in you forked repo. 
Did you push your local commit to upstream ('git push') ?

frantisadlo, 2024-05-05:
How about now? I see it in my account.

Oh, I am not so clear, again. Sorry for that. So:
45352-45358 are already there in 0.3.3 and they are setting parameters for Inverter limits.
I have been added 47900 - 47917 section which is read directly from BMS and can be different (my issue and solved with it).

As minor I add Load regulation as well, from 47595-47599.

Frantisek

mletenay, 2024-05-05:
Aaaaa, I just realized we are talking about ILLEGAL DATA ADDRESS in read_settings_data() and not in read_runtime_data() !
I forgot I added such method there at all, since HA is not using it and I did not test it for quite some time.
Yes, it indeed fails on ET fw1023.
Fortunately, settings are requested one by one, so we do not have to filter them during setup, but lazily on request.

I see now what you are trying to achieve !

mletenay, 2024-05-05:
OK, so I took your changes, did some few tweaks in names to make it consistent with rest.
Added also BMS2 settings for ET 25K+ inverters.
Improved loading of settings so it will not fail when setting is not supported and will remove it from list.
(You can revert your commit and pull my changes (git rever 2a161f3; git pull --rebase).

closed 2024-05-05 by mletenay

frantisadlo, 2024-05-07:
Good morning, 
  thank you for all. I aplied it works  good.
Here is just one small question: It is in ET.py file.
Are those folowing two IDs on lines 390-392 specificaly meant to use with hyphen in naming or is it misstyping and could be use with underscore
`bms_bat_warning-code ` , `bms_bat_alarm-code?` I am using those names as variables and it is not allowed to use it in python and I will have to do some extra mapping. I have changed to underscore for now.
`Long("bms_bat_warning_code", 47911, "BMS Battery Warning Code"),`
`# Reserved`
`Long("bms_bat_alarm_code", 47913, "BMS Battery Alarm Code"),`