marcelblijleven/goodwe #78: info flashing in and out

oziee, 2024-05-13:
I have a SBP5000 and in HA the data flashes in with info and then out with unknown over and over..

the logs ouput
```
2024-05-13 16:47:51.719 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback _SelectorDatagramTransport._read_ready()
Traceback (most recent call last):
File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
self._context.run(self._callback, *self._args)
File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1238, in _read_ready
self._protocol.datagram_received(data, addr)
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 49, in datagram_received
if self.command.validator(data):
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 148, in <lambda>
lambda x: self._validate_response(x, response_type),
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 168, in _validate_response
logger.debug("Response has unexpected length: %d, expected %d.", len(data), data[6] + 9)
~~~~^^^
IndexError: index out of range
2024-05-13 16:48:43.049 ERROR (MainThread) [custom_components.goodwe.coordinator] Error fetching GoodWe data: ('No valid response received even after 10 retries', 3)
```

any help?

mletenay, 2024-05-19:
Your inverter is of "ES" kind.
You seem to receive lots of incomplete/fragmented responses, it is rare but unfortunately it happens.
I have just implemented an attempt to recovers from these fragmented packets (if that is actually what happens).
Please, update to latest version and try again.

oziee, 2024-05-20:
@mletenay cheers.. thanks for trying..

seems to get some data but then nothing else ever comes in.. just sits there
```
DEBUG:asyncio:Using selector: KqueueSelector
DEBUG:goodwe:Connecting to ES family inverter at 10.0.1.51:8899.
DEBUG:goodwe.protocol:Creating lock instance for current event loop.
DEBUG:goodwe.protocol:Sending: aa55c07f0102000241
DEBUG:goodwe.protocol:Received response fragment: aa557fc001824d3130313043475735303030532d425023313000000000000000000000000000393530303042505332323357303138
```
waited..waited and waited more but nothing else ever logged..

I still get `index out of range` errors or the test just sit there and do nothing

```
python inverter_check.py
2024-05-20 14:57:19,352 __init__(54) - DEBUG: Using selector: KqueueSelector
2024-05-20 14:57:19,352 connect(52) - DEBUG: Connecting to ES family inverter at 10.0.1.51:8899.
2024-05-20 14:57:19,352 _ensure_lock(58) - DEBUG: Creating lock instance for current event loop.
2024-05-20 14:57:19,353 _send_request(176) - DEBUG: Sending: aa55c07f0102000241
2024-05-20 14:57:19,742 default_exception_handler(1771) - ERROR: Exception in callback _SelectorDatagramTransport._read_ready()
handle: <Handle _SelectorDatagramTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/asyncio/selector_events.py", line 1169, in _read_ready
    self._protocol.datagram_received(data, addr)
  File "/Users/oziee/Downloads/goodwe-master 4/goodwe/protocol.py", line 131, in datagram_received
    if self.command.validator(data):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/oziee/Downloads/goodwe-master 4/goodwe/protocol.py", line 487, in <lambda>
    lambda x: self._validate_response(x, response_type),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/oziee/Downloads/goodwe-master 4/goodwe/protocol.py", line 507, in _validate_response
    raise PartialResponseException(len(data), data[6] + 9)
                                              ~~~~^^^
IndexError: index out of range
```

mletenay, 2024-05-21:
Fixed in v0.4.5

closed 2024-05-21 by mletenay

oziee, 2024-05-21:
Thanks.. I'll give it a try tonight