marcelblijleven/goodwe #82: reading only single sensor values

Wiese66, 2024-07-03:
Hello,

thanks for these great library.
As I´m not into python very much, I haven´t found the solution myself.

I want to read only single sensors (i.e. "ppv" or "active_power_total") as I normaly don´t need all sensor data at once. I hope that I can reduce a bit the stress on the bus as currently all modbus values are read on very call to read_runtime_data, as far as I understood the code.
I found the function for "read_setting" but not an equivalent for the current sensor values. Would it be very complex to add such a function? Or would there no effect on reading the modbus? I have an ET15k, if this matters.

Best regards
  Chris

mletenay, 2024-07-04:
Well, at this moment it is not possible to read single "sensor", only all of them.
But I did make necessary changes some time ago, so it should not be difficult to implement it now (at least for ET and DT inverters).
Give me a couple of days, I'll implement it.

Wiese66, 2024-07-04:
Thanks a lot and no hurry, I appriciate it very much :)

mletenay, 2024-07-05:
There's new method read_sensor, let's try if it works for you ...

Wiese66, 2024-07-06:
So this were some fast couple of days :)
I´ve done some quick tests and this works like a charm. As you will know, the read of calculated values (i.e. ppv) will not work but for me this is not needed at all.
So thank you very much!

closed 2024-07-06 by Wiese66