marcelblijleven/goodwe #83: "No valid response received even after 3 retries" with GW4200D-NS

haarts, 2024-07-10:
I'm just running the example script from the README (obviously with the IP address changed to point to my inverter) and I'm getting "No valid response received even after 3 retries".

The Wifi module has firmware version V1.2.1.20(Boot:V1.1.4).
The DSP firmware version is 12.00.12.
The ARM firmware version is 24.271 (updated OTA by Goodwe today).

The SEMS/SolarGo/HA integrations work as expected.

I know this lib uses UDP so I tried a couple of times to no avail. 

What I did notice is that wifi connection is somewhat unstable. I had two disconnects today, the Wifi module seems to revert to AP mode. Not sure this is related.

Is there something I can try?

IpPsar01, 2024-07-11:
For what it's worth...

I have the same inverter and after many hours I have just gotten to the point that I get a response using a slightly changed example script:
After the last import line add:
import logging
import sys

logging.basicConfig(
    format="%(asctime)-15s %(funcName)s(%(lineno)d) - %(levelname)s: %(message)s",
    stream=sys.stderr,
    level=getattr(logging, "DEBUG", None),
)

IP_ADDRESS = "your inverter IP"
FAMILY = "NS"
COMM_ADDR = 0x7f
PORT = 8899
TIMEOUT = 1
RETRIES = 30

And I changed the goodwe.connect call to: 
          inverter = await goodwe.connect(IP_ADDRESS, FAMILY, COMM_ADDR, TIMEOUT, RETRIES)

When running this repetitively I get many many failures but once every 50 to a 100 attempts it works. It first connects and than tries to get the actual data which sometimes works. So at least I know there is (sometimes) a correct response from the inverter. 

Perhaps I open a question on how get less failures because I seem to get stuck at that. Running a large amounts of pings to the inverter gives 0% loss!

Hope it helps.

haarts, 2024-07-18:
That seems unworkable. I have my Wifi drop entirely often too. Support is on the case but it's hard to get an answer other than 'have you tried turning it off an on again?'. 
Alas. Thanks for the help!

closed 2024-07-18 by haarts

Dale-Muccignat, 2024-08-14:
Heya, did you end up finding a solution to this? I have a GW3000D-NS and can't get any response using the script

haarts, 2024-08-15:
I did not.

-------- Original Message --------

chatz86, 2024-09-15:
Hey. I have a gw5000d-ns. Was having some issues which have been rectified. 

My DSP FW version: 04.00.04.--
ARM FW: 24.271

My DSP version is different to yours I notice. Perhaps start there and contact support. 

It was in the too hard basket for the first port of communication however the 2 guys I got after that were absolutely  amazing!
[image]