marcelblijleven/goodwe #88: Question: Availability of Data to Query

madtempest, 2024-08-06:
Hi,

I wanted to understand more regarding the available data that is available to query using this library.

For context: I have a GW9.9KAU-DT (aka SDT or DT family) with the WiFiKit v1.5 adapter and the GM3000 smart meter in the meter box connected via an ethernet cable (I know this is using RS-485 in the background).

I can connect using this library to port 8899/UDP and when I run the inverter_check.py script I get back:

Identified inverter
- Model:
- SerialNr: 
- Rated power: 0
- A/C output type: None
- Firmware: 20.20.38
- ARM firmware: None
- Modbus version: None
- DSP1 version: 20
- DSP2 version: 20
- DSP svn version: 1243
- Arm version: 56
- ARM svn version: 124

2024-08-06 14:07:06,817 __init__(54) - DEBUG: Using selector: EpollSelector
2024-08-06 14:07:06,817 _ensure_lock(59) - DEBUG: Creating lock instance for current event loop.
2024-08-06 14:07:06,818 _send_request(212) - DEBUG: Sending: READ 73 registers from 30100 (7f0375940049d5c2)
2024-08-06 14:07:07,077 datagram_received(153) - DEBUG: Received: aa557f039218050613393910740018109f0024ffffffffffffffffffffffffffffffff106510601054097b095d0981002200230023138313831384000009ad000100000000000000000000ffffff860000000003e7ffff016effffffff00de00001b500000012000200000000000000000ffffffffffffffffffffffffffffffff010a18ce0c8100000020000004db010a0004000000654071
2024-08-06 14:07:07,078 connection_lost(138) - DEBUG: Socket closed.
2024-08-06 14:07:07,078 _send_request(212) - DEBUG: Sending: READ register 30196 (7f0375f40001d5ea)
2024-08-06 14:07:08,079 _timeout_mechanism(223) - DEBUG: Failed to receive response to READ register 30196 (7f0375f40001d5ea) in time (1s).
2024-08-06 14:07:08,080 connection_lost(138) - DEBUG: Socket closed.
2024-08-06 14:07:08,080 _send_request(210) - DEBUG: Sending: READ register 30196 (7f0375f40001d5ea) - retry #1/3
2024-08-06 14:07:09,067 validate_modbus_rtu_response(206) - DEBUG: Response is command failure: ILLEGAL DATA ADDRESS.
2024-08-06 14:07:09,067 datagram_received(167) - DEBUG: Received exception response: aa557f8302a0e9
2024-08-06 14:07:09,068 read_runtime_data(204) - INFO: Meter values not supported, disabling further attempts.
2024-08-06 14:07:09,068 connection_lost(138) - DEBUG: Socket closed.
timestamp:               Timestamp = 2024-05-06 19:57:57
vpv1:            PV1 Voltage = 421.2 V
ipv1:            PV1 Current = 2.4 A
ppv1:            PV1 Power = 1011 W
vpv2:            PV2 Voltage = 425.5 V
ipv2:            PV2 Current = 3.6 A
ppv2:            PV2 Power = 1532 W
ppv:             PV Power = 2543 W
vline1:                  On-grid L1-L2 Voltage = 419.7 V
vline2:                  On-grid L2-L3 Voltage = 419.2 V
vline3:                  On-grid L3-L1 Voltage = 418.0 V
vgrid1:                  On-grid L1 Voltage = 242.7 V
vgrid2:                  On-grid L2 Voltage = 239.7 V
vgrid3:                  On-grid L3 Voltage = 243.3 V
igrid1:                  On-grid L1 Current = 3.4 A
igrid2:                  On-grid L2 Current = 3.5 A
igrid3:                  On-grid L3 Current = 3.5 A
fgrid1:                  On-grid L1 Frequency = 49.95 Hz
fgrid2:                  On-grid L2 Frequency = 49.95 Hz
fgrid3:                  On-grid L3 Frequency = 49.96 Hz
pgrid1:                  On-grid L1 Power = 825 W
pgrid2:                  On-grid L2 Power = 839 W
pgrid3:                  On-grid L3 Power = 852 W
total_inverter_power:            Total Power = 2477 W
work_mode:               Work Mode code = 1
work_mode_label:                 Work Mode = Normal
error_codes:             Error Codes = 0
warning_code:            Warning code = 0
apparent_power:                  Apparent Power = 0 VA
reactive_power:                  Reactive Power = -122 var
power_factor:            Power Factor = 0.999
temperature:             Inverter Temperature = 36.6 C
e_day:           Today's PV Generation = 22.2 kWh
e_total:                 Total PV Generation = 699.2 kWh
h_total:                 Hours Total = 288 h
safety_country:                  Safety Country code = 32
safety_country_label:            Safety Country = 50Hz 230Vac Default
funbit:                  FunBit = 266
vbus:            Bus Voltage = 635.0 V
vnbus:           NBus Voltage = 320.1 V
derating_mode:           Derating Mode code = 32
derating_mode_label:             Derating Mode = Export power derating

I don't see anything here that would remotely suggest anything related to current load or what is being exported and what is not. Am I right in assuming that this kind of information is not available to me in my circumstance without either a firmware upgrade or changing the WiFiKit to a v2 module instead?

Thank you for this awesome library!

mletenay, 2024-08-06:
No, not really.
WifiKit v2 will not change anything, that is only communication gateway/protocol to access inverter's internal modbus registers where the data are stored. Firmware update will neither change anything.

The values you see are all publicly known registers of DT familiy inverters.
There might be others somewhere there, but they are not documented/known.

closed 2024-08-06 by mletenay

madtempest, 2024-08-06:
Thank you for your quick response. It's appreciated!

nicve, 2024-08-06:
Hello. Today I have shared some additional documentation in the #general channel on discord. Perhaps this will help? It was supplied today by GoodWe. @mletenay