marcelblijleven/goodwe #97: Use modes TCP on Port 502

pme101, 2025-01-14:
Hello,

I have a Goodwe GW20K-ET with the Wifi/LAN Kit 2.0. Modbus TCP seems to work correctly on the inverter (at least I can successfully read via Modbus TCP on port 502 with the following script: https://www.photovoltaikforum.com/thread/237857-modbus-scanner-und-werkzeuge-f%C3%BCr-goodwe-wechselrichter/?postID=4014428#post4014428).

According to the documentation of this package, Modbus TCP should be supported since v0.4.x. I have version 0.4.8 of the goodie package and use Python 3.13.x.

When I try to run the test script (with the correct IP address of the inverter), I get the following error:

`goodwe.exceptions.InverterError: Unable to connect to the inverter at host=192.168.68.63, or your inverter is not supported yet.
Failures=[RequestFailedException("No valid response received to 'aa55c07f0102000241' request."), RequestFailedException("No valid response received to 'f70388b800213ac1' request.", 1), RequestFailedException("No valid response received to '7f03753100280409' request.", 1), RequestFailedException("No valid response received to 'aa55c07f0102000241' request.", 1)]`

Any idea what could be causing my issue.

Thanks!

Chris56, 2025-04-20:
Your code does not access the modbus port 502 but the udp port 8899.
8899 is the port the SolarGo app uses.
My observation is, that after SolarGo has accessed the inverter, for a couple of hours the test.py gets answers from udp port 8899. After that it refuses the connections until SolarGo again initializes a connection.
I've not yet had the time to trace the SolarGo initialization.

mgx0, 2025-07-10:
what shall i do to enable modbus port 502 please?

alenpaulvarghese, 2026-08-30:
## Same root cause, confirmed with fresh diagnostics (GW5K-DNS-G40, DT family)

I hit the exact same problem with a different inverter: **GW5K-DNS-G40** (DT family) using the **WLA0000-01-00P v2.0 LAN module**. `goodwe.connect(host)` fails, but `connect(host, port=502)` works perfectly. I traced the full failure path — the library's `discover()` never tries TCP/502 when all UDP/8899 probes fail.

### Diagnostic evidence (against live inverter at `192.168.0.40`)

| Probe | Result |
|---|---|
| AA55 discovery on UDP 8899 | no response |
| Modbus RTU over UDP 8899 (comm_addr 0xF7 / 0x7F / 0x01 / 0x00) | no response |
| **Modbus/TCP on port 502** | **SUCCESS — `GW5K-DNS-G40`, S/N `55000NBA263L1473`** |
| UDP broadcast on 48899 (`WIFIKIT-214028-READ`) | responds with new dongle format |

`python -c` reproducer:

```python
import asyncio, goodwe

async def main():
    # Fails even though the inverter is fully reachable:
    inv = await goodwe.connect("192.168.0.40", retries=3)   # → InverterError
    # Works only when port is forced:
    inv2 = await goodwe.connect("192.168.0.40", port=502, retries=3)  # → OK

asyncio.run(main())
```

### Root cause

`discover()` in `goodwe/__init__.py` only probes UDP 8899 (AA55 discovery + per-family ET/DT/ES), then gives up and raises `InverterError`. It **never falls back to Modbus/TCP on port 502**, even though the README states port 502 is supported since v0.4.x for the v2.0 dongle.

### Working fix (small code change)

Add a TCP/502 fallback loop at the end of `discover()`, after the UDP probes fail:

```python
async def discover(
    host: str, port: int = GOODWE_UDP_PORT, timeout: int = 1, retries: int = 3
) -> Inverter:
    failures = []

    if port == GOODWE_UDP_PORT:
        # ... existing AA55 UDP discovery ...
        try:
            ...
        except InverterError as ex:
            failures.append(ex)

    # ... existing per-family UDP probes ...

    # NEW: Fallback to Modbus/TCP on port 502 for v2.0 LAN modules (WLA0000-01-00P)
    if port == GOODWE_UDP_PORT:
        logger.debug("UDP probes failed, trying Modbus/TCP on port 502.")
        for inv in [ET, DT, ES]:
            i = inv(host, GOODWE_TCP_PORT, 0, timeout, retries)
            try:
                await i.read_device_info()
                await i.read_runtime_data()
                return i
            except InverterError as ex:
                failures.append(ex)

    raise InverterError(
        "Unable to connect to the inverter at "
        f"host={host}, or your inverter is not supported yet.\n"
        f"Failures={str(failures)}"
    )
```

(`GOODWE_TCP_PORT` is already defined in `goodwe/const.py`; the inverter protocols already pick `TcpInverterProtocol` automatically for port 502.)

I have this locally working — `connect("192.168.0.40")` now succeeds with no manual port. Happy to open a PR if it helps. Thanks!