marcelblijleven/goodwe #107: Different data returned when using Port 8899 vs Port 502

DF338, 2025-05-02:
When I run test.py using Port 8899 (default) I get the correct result for the Active Power register (offset 30196) but I get -1 (or zero) if I specify Port 502.  I've uploaded the two result files.  They are taken a couple of minutes apart so other values have changed, but the (Port 502) result always remains -1.  In both cases, the returned data is said to come from Register offset 30916 (see below)

 PowerS(id_='active_power',
        offset=30196,
        name='Active Power',
        size_=2,
        unit='W',
        kind=<SensorKind.GRID: 5>))

My inverter is a GW10K-MS-30 and the associated Smartmeter is a Goodwe GMK110.

The only other noticable difference when running test.py against Port 502 vs Port 8899 is that the results come back much quicker when I specify Port 502 and there is a wait of approx 7 seconds when running it using the default 8899 port

[Using Port 502.txt](https://github.com/user-attachments/files/20007027/Using.Port.502.txt)
[Using Port 8899.txt](https://github.com/user-attachments/files/20007028/Using.Port.8899.txt)

sunnysingh-84, 2025-05-23:
What dongle are you using with your inverter?

active_power is the amount of power your importing/exporting from grid, re-check the results during the middle of the day when your making more solar to see what active_power reading you get?

DF338, 2025-05-23:
Thanks for your comment - I'm using the V2 wifi/LAN dongle (modbus supporting) and have tested both night and day.   The issue seems to be with the sensor number in the library.  If I change the sensor No. in the library source code to 30195 (0x75F3) then using Port 502 gives the correct answer but Port 8899 gives zero.  When the sensor is set to 30196 (0x75f4) [as is is in the un-modified library], then port 8899 gives the correct answer and port 502 gives zero.  So it seems that the active_power sensor number is being interpreted differently in the two different program paths that port 8899 (UDP) and port 502 (tcp) are following..
Cheers

mletenay, 2025-05-23:
Well, that is not possible. TCP and UDP are just different transport layers, the data retrieval/mapping code is exactly the same.

DF338, 2025-05-23:
Hi there.  This is the code I modified in dt.py:

def __init__(self, host: str, port: int, comm_addr: int = 0, timeout: int = 1, retries: int = 3):
        super().__init__(host, port, comm_addr if comm_addr else 0x7f, timeout, retries)
        self._READ_DEVICE_VERSION_INFO: ProtocolCommand = self._read_command(0x7531, 0x0028)
        self._READ_RUNNING_DATA: ProtocolCommand = self._read_command(0x7594, 0x0049)
        self._READ_METER_DATA: ProtocolCommand = self._read_command(0x75f4, 0x01)  **#DF changed to 0x75F3 for Port 502**
        self._sensors = self.__all_sensors
        self._sensors_meter = self.__all_sensors_meter
        self._settings: dict[str, Sensor] = {s.id_: s for s in self.__all_settings}
        self._has_meter: bool = True

When I make this change, I get the correct answer for active_power if I specify Port 502 but zero if I specify Port 8899 (and vice versa)
Cheers, DF

DF338, 2025-05-26:
I've added one more piece of info from my setup that might be useful.  If I scan the registers using Modbus Poll (TCP Port 502) I get the attached scan.  In this , Register 30195 is showing the correct data for active_power 

[image]