marcelblijleven/goodwe #108: Delayed Charging mode, battery current limits and other new functions in SolarGo

piomar123, 2025-05-22:
There are a few new functions added recently in SolarGo app that would be interesting to have in the library:

### Delayed charging:

[image]

[image]

### Off-grid SOC recovery:

[image]

### Battery current limits

[image]

I can try to capture the data sent by SolarGo to the inverter but I wanted to ask what is the best option to do it?
Is Wireshark the best option or maybe the library has any built-in listening mode?

sunnysingh-84, 2025-05-23:
Delayed Charging is also known as Smart Charging.  I use this feature daily to stop my batteries charging from PV and instead use Fast Charge during off peak which is when my utility provider offers free electricity.  Smart Charging is partially coded into the library but its not written into the home assistant integration as of yet. 

From memory, my work around for using this in home assistant was as follows:

1. Enable Smart Charging (Delayed Charging) via the SolarGo app as following:

Monthly-Repeat: Every Month
Peak Power Sales Limit: 100%
PV Priortise Charging Battery: On
Start Charging Time: 23:59 

2. Update /homeassistant/custom_components/goodwe/switch.py to include the following 

    GoodweSwitchEntityDescription(
        key="smart_charging_switch",
        translation_key="smart_charging_switch",
        device_class=SwitchDeviceClass.SWITCH,
        setting="smart_charging_enable",
        polling_interval=30,
    ),

3. Update /homeassistant/custom_components/goodwe/translations/en.json to include the following

            "smart_charging_power": {
                "name": "Smart Charging Power"

4: Update /homeassistant/custom_components/goodwe/strings.json

      "smart_charging_switch": {
        "name": "Smart charging switch"

5. Add the following into /homeassistant/configuration.yaml

modbus:
  - name: goodwe_tcp
    type: tcp
    host: <GOODWE DONGLE IP ADDRESS>
    port: 502
    delay: 1
    message_wait_milliseconds: 25
    timeout: 5
    
    sensors:
      - name: smart_charging_power
        slave: 247
        address: 47592
        scan_interval: 0
        scale: 0.1
        min_value: 0 
        max_value: 100
        unit_of_measurement: "%"

thisIsLiving, 2026-09-12:
> As an alternative to Step 5 that [@sunnysingh-84](https://github.com/sunnysingh-84) suggested, I modified
> 
> `/homeassistant/custom_components/goodwe/number.py`
> 
> to include the following code under `NUMBERS = ( ...`
> 
> ```
>     GoodweNumberEntityDescription(
>         key="smart_charging_power_limit",
>         translation_key="smart_charging_power_limit",
>         entity_category=EntityCategory.CONFIG,
>         native_unit_of_measurement=PERCENTAGE,
>         native_step=1,
>         native_min_value=0,
>         native_max_value=100,
>         # The inverter stores this setting in tenths (e.g. 760 => 76.0%).
>         # Use the modbus raw-register API (with hyphen) and scale on read/write.
>         getter=lambda inv: inv.read_setting("modbus-47592"),
>         mapper=lambda v: int(v / 10) if v is not None else 0,
>         setter=lambda inv, val: inv.write_setting("modbus-47592", int(val) * 10),
>             filter=lambda inv: True,
>     ),
> ```
> 
> Restart Home Assistant and it should show up now as a configurable entity

I've updated my inverter firmware and GoodWe must have changed the register location. With the help of Claude the new register for "Peak Power Sales Limit" is 47837, tested and working.

PiotrMarcinczyk-TomTom, 2026-09-26:
🤖: Answering my own question from the top, and adding what I found.

**How to capture SolarGo's traffic:** [PCAPdroid](https://github.com/emanuele-f/PCAPdroid) on Android (no root), with *Target apps* = SolarGo and *Traffic dump* = PCAP file. SolarGo's UDP discovery (`WIFIKIT-214028-READ` broadcast to port 48899) doesn't get through PCAPdroid's VPN. So **connect SolarGo to the inverter first, then start the capture**. The `.pcap` then contains plain Modbus-RTU over UDP 8899 (requests `F7 03/06/10 …`, replies `AA 55 F7 …`). Tested on a GW8KN-ET, DSP 12.197 / ARM 31.

**Off-grid SOC recovery = register 45287** (SoC %). Changing 40 → 50 → 40 in SolarGo:
```
08:34:40  WRITE 45287 = 50   ack   f706b0e700328a7e
08:34:45  WRITE 45287 = 40   ack   f706b0e700280bb5
```
It's not in the 2022 ET Modbus map. PR to follow.

**Battery current limits = 45353 (charge) / 45355 (discharge)**, 0.1 A units. They're already in the library as `battery_charge_current` / `battery_discharge_current`, and `write_setting()` encodes them the way SolarGo does:
```
08:35:24  WRITE 45353 = 185  ack   (18.5 A)
08:35:40  WRITE 45355 = 0    ack   (0 A -> battery discharge stops)
```
The map describes them as inverter-side limits. The BMS limits (47900+) still apply, so a value above the BMS maximum is capped by the BMS.

**Delayed charging = Smart charging:** register dump only so far. It matches what @sunnysingh-84 and @thisIsLiving found: `47609` = Smart Charging enable, and the settings are in block "group 8" at `47589–47594` (start/end time, weekday bits, `47592` = peak power sales limit in 0.1 %, e.g. 1000 = 100 %). The library reads the same block as `peak_shaving_mode` (47589), so writing peak shaving overwrites the delayed-charging configuration (see #139). I'll confirm with a capture of toggling it in SolarGo.

**Two side findings from the same captures (PRs to follow):**
- SolarGo's *restart inverter* writes `361` to the undocumented `45221` (`f706b0a501696a01`). The documented `45220` only reconnects to the grid.
- SolarGo **writes the phone's local time to the inverter clock (45200) every time it connects**. That explains clock jumps people may put down to NTP.

piomar123, 2026-09-26:
🤖: Answering my own question from the top, and adding what I found.

**How to capture SolarGo's traffic:** [PCAPdroid](https://github.com/emanuele-f/PCAPdroid) on Android (no root), with *Target apps* = SolarGo and *Traffic dump* = PCAP file. SolarGo's UDP discovery (`WIFIKIT-214028-READ` broadcast to port 48899) doesn't get through PCAPdroid's VPN. So **connect SolarGo to the inverter first, then start the capture**. The `.pcap` then contains plain Modbus-RTU over UDP 8899 (requests `F7 03/06/10 …`, replies `AA 55 F7 …`). Tested on a GW8KN-ET, DSP 12.197 / ARM 31.

**Off-grid SOC recovery = register 45287** (SoC %). Changing 40 → 50 → 40 in SolarGo:
```
08:34:40  WRITE 45287 = 50   ack   f706b0e700328a7e
08:34:45  WRITE 45287 = 40   ack   f706b0e700280bb5
```
It's not in the 2022 ET Modbus map. PR to follow.

**Battery current limits = 45353 (charge) / 45355 (discharge)**, 0.1 A units. They're already in the library as `battery_charge_current` / `battery_discharge_current`, and `write_setting()` encodes them the way SolarGo does:
```
08:35:24  WRITE 45353 = 185  ack   (18.5 A)
08:35:40  WRITE 45355 = 0    ack   (0 A -> battery discharge stops)
```
The map describes them as inverter-side limits. The BMS limits (47900+) still apply, so a value above the BMS maximum is capped by the BMS.

**Delayed charging = Smart charging:** register dump only so far. It matches what @sunnysingh-84 and @thisIsLiving found: `47609` = Smart Charging enable, and the settings are in block "group 8" at `47589–47594` (start/end time, weekday bits, `47592` = peak power sales limit in 0.1 %, e.g. 1000 = 100 %). The library reads the same block as `peak_shaving_mode` (47589), so writing peak shaving overwrites the delayed-charging configuration (see #139). I'll confirm with a capture of toggling it in SolarGo.

**Two side findings from the same captures (PRs to follow):**
- SolarGo's *restart inverter* writes `361` to the undocumented `45221` (`f706b0a501696a01`). The documented `45220` only reconnects to the grid.
- SolarGo **writes the phone's local time to the inverter clock (45200) every time it connects**. That explains clock jumps people may put down to NTP.

piomar123, 2026-09-26:
🤖: Follow-up with the capture of **Delayed charging** in SolarGo (same PCAPdroid method as above, GW8KN-ET DSP 12.197 / ARM 31). Enabling it with peak power sales limit 100 % and time 23:59:

```
15:30:40  WRITE_MULTI 47589..47594 = 0000 173b fa7f 03e8 0000 0fff
15:30:42  WRITE 47613 = 0
15:30:42  WRITE 47000 = 0      work mode -> General (self-use)
15:30:42  WRITE 47533 = 1      clearECOtime
15:30:43  WRITE 47609 = 1      smart charging enable
```

Disabling it again:

```
15:40:54  WRITE 47000 = 0
15:40:54  WRITE_MULTI 47511..47512 = 0001 0000   EMS mode AUTO, power 0
15:40:55  WRITE 47533 = 1
15:40:55  WRITE 47591 = 895    (0x037F)
15:40:55  WRITE 47609 = 0
```

So Delayed charging = Smart charging, as suspected:

- `47609` enables it.
- Group 8: `47589` start time, `47590` end time (hh:mm as hi/lo byte), `47591` high byte = function (`0xFA` smart charge enabled; SolarGo writes `0x03` when disabling, which the map lists as "peak shaving disabled"), low byte = weekdays; `47592` = peak power sales limit in 0.1 % (1000 = 100 %); `47593`/`47594` written as `0` / `0x0FFF` (meaning unknown).
- SolarGo switches the work mode to General (`47000 = 0`) and writes `clearECOtime` (`47533 = 1`, "clear all economical mode settings").

Side effect worth knowing: after this, the eco mode slots were **disabled** (their on/off byte went `0xFF` → `0x00`; times and powers kept). No write to them was captured, so it's the `clearECOtime` effect. Switching back to Eco in SolarGo writes `47000 = 3` and `47511..47512 = 1, 0` but doesn't re-enable any slot - each one has to be switched on again by hand.

Because group 8 is shared with `peak_shaving_mode`, writing one overwrites the other (#139).