marcelblijleven/goodwe #118: Support for Inverter GW8.6K-EHB

Zathras42, 2025-09-17:
Hello,

Please add support for EHB series inverters.
Model number is GW8.6K-EHB
SN starts 98600NAF255L....

Unit has 4 MPPT's

Zathras42, 2026-02-01:
No luck for me.
`H:\Projects\Smart Home\! Home Assistant\Solar\goodwe>python test.py
Traceback (most recent call last):
  File "H:\Projects\Smart Home\! Home Assistant\Solar\goodwe\test.py", line 16, in <module>
    asyncio.run(get_runtime_data())
  File "D:\Programs\Python\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "D:\Programs\Python\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "H:\Projects\Smart Home\! Home Assistant\Solar\goodwe\test.py", line 8, in get_runtime_data
    inverter = await goodwe.connect(ip_address)
  File "H:\Projects\Smart Home\! Home Assistant\Solar\goodwe\goodwe\__init__.py", line 57, in connect
    return await discover(host, port, timeout, retries)
  File "H:\Projects\Smart Home\! Home Assistant\Solar\goodwe\goodwe\__init__.py", line 143, in discover
    raise InverterError(
goodwe.exceptions.InverterError: Unable to connect to the inverter at host=192.168.42.51, or your inverter is not supported yet.
Failures=[MaxRetriesException(), RequestFailedException('No valid response received even after 3 retries', 1), RequestFailedException('No valid response received even after 3 retries', 1), RequestFailedException('No valid response receivedeven after 3 retries', 1)]`

I'm also using the Home Assistant GoodWe Inverter integration which does get most data except for MPPT 3 & 4.

shanesmith1956-ai, 2026-09-10:
GW5K-EHB-A confirmed working over Modbus TCP with goodwe 0.4.10

I can confirm successful local communication with a GoodWe GW5K-EHB-A using goodwe 0.4.10.

This test arose because our Home Assistant GoodWe integration had been unable to obtain local inverter data for several weeks, despite extensive troubleshooting of the inverter, LAN connection, firmware and Modbus TCP availability.

To isolate the problem, we tested the inverter directly with the goodwe Python library, independently of Home Assistant and SEMS+.

Environment

Model: GW5K-EHB-A
Serial prefix: 55000NAF256L...
Firmware: 04094-01-S10
ARM firmware: 02041-34-S00
Modbus version: 2
TCP port: 502
Family: EH
Modbus address: 247
Python: 3.14.6

Connection succeeded using:

inverter = await goodwe.connect(
    "192.168.x.x",
    port=502,
    family="EH",
    comm_addr=247,
    timeout=2,
    retries=1
)

The inverter correctly returned its model, serial, firmware and Modbus version.

read_runtime_data() also succeeded and returned valid PV, battery, load, grid, meter and BMS data.

Example values from one test:

Battery SOC: 69%
PV power: 443 W
House consumption: 1328 W
Battery modules detected: 6
Battery SOH: 100%
Meter communication status: 1

The readings were consistent with the corresponding SEMS+ values at the time.

This confirms that, at least for this GW5K-EHB-A / NAF-series inverter, local Modbus TCP on port 502 with address 247 is working successfully with goodwe 0.4.10.

The result also suggests that our remaining Home Assistant issue lies above the basic inverter/LAN/Modbus communication layer.

Happy to provide additional read-only register output or testing if it would assist EHB support or help identify any Home Assistant integration compatibility issue.

Zathras42, 2026-10-15:
Apparently issue was fixed and shipped with HA core 2026.5 (https://github.com/home-assistant/core/pull/166809) and updated GoodWe library recognizes EHB inverters as 4 MPPT models.

Yet HA still does not show all 4 MPPTs.  Issue I raised (https://github.com/orgs/home-assistant/discussions/966) has been closed as resolved.
I can't see how it's closed when it simply doesn't work...